commaai/openpilot #21145: DM doesn't detect if my eyes are closed, or asleep

zorrobyte, 2021-06-04:
While DM does a great job badgering me if I'm distracted, DM doesn't seem to care if I'm sleeping or passed out. Next time you are in a safe environment, try closing your eyes and pretending you are sleeping, or worse, passed out.

Good DM would help prevent crashes such as:
https://youtu.be/lssmh4GH188

zorrobyte, 2021-06-04:
Sat in my car in various poses with my eyes closed. Received the "Keep eyes on road" alert but never a disengagement:
d5b9bf105ad64d5d|2021-06-04--18-46-24

Maybe timeout is just super high? Feels like something isn't working properly with how quick DM chastises me for looking at my phone, head unit. I'm happy to replicate further and go on longer "drives".

ZwX1616, 2021-06-05:
Hmm the alerts don't go beyond green if the car is at standstill.

zorrobyte, 2021-06-05:
Feature not bug, closing

closed 2021-06-05 by zorrobyte