commaai/openpilot #21149: Corolla Hatchback unrecognized on latest master-ci

y-a-n-w-a-n-g, 2021-06-05:
**Describe the bug**

My Corolla hatchback '20 used to work fine with openpilot. I've been using master-ci for all of versions of 0.8 and have been updating it almost weekly. After updating today, I've been getting car not supported messages and can only run dashcam. After restarting and checking the cables a few times, I gave up and uninstalled openpilot from the menu. After it rebooted, I reinstalled openpilot using the regular url, openpilot.comma.ai, and it's working again. I did the whole uninstall and reinstall without unplugging my device, so it's not the cables. I think some changes on master-ci messed up the devices

**How to reproduce or log data**

Update to the latest version of master-ci and attempt to run openpilot on a Corolla Hatchback 2020.

**Expected behavior**

Openpilot should run fine.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: f28c43ff6a3dea6f
 - Route: f28c43ff6a3dea6f|2021-06-04--19-18-15 (I think)
 - Timestamp: Entire ride
 - Version: master-ci 0.8.5?
 - Car make/model: Toyota Corolla Hatchback 2020

**Additional context**

N/A

y-a-n-w-a-n-g, 2021-06-05:
Also in useradmin, it shows the platform as `mock`  when on `master-ci`. After reinstalling, on `release2` it shows `TOYOTA COROLLA TSS2 2019`, which is the wrong year but still works.

ShaneSmiskol, 2021-06-05:
Make sure all your connections are seated properly, I can't see any firmware fingerprints from any of your recent drives. Especially your comma power v2 near your obd2 port; that it's seated and that the ethernet cable is plugged in properly.

y-a-n-w-a-n-g, 2021-06-05:
I unplugged and replugged the cables, and they seem to be seated in. Tried driving and for the latest drive the carFw is empty in the carParams log. Are there any other next steps I should do?

VirtuallyChris, 2021-06-08:
If you're in warranty, email support@comma.ai so we can assist :)

y-a-n-w-a-n-g, 2021-06-10:
I've already sent the support email, but just to leave an update, I swapped ethernet cables and I still got `carFw = []`.

adeebshihadeh, 2021-06-10:
Closing as this sounds like a hardware issue. Support will take care of this.

closed 2021-06-10 by adeebshihadeh