commaai/openpilot #21156: update_requirements needed sudo apt install libcurl4-gnutls-dev

ClockeNessMnstr, 2021-06-05:
Operating system: [e.g. Ubuntu 20.04]

on a fresh install I cloned into openpilot and ran ./update_requirements. 
it instructed to run the following + path edits. 

sudo apt-get update; sudo apt-get install --no-install-recommends make build-essential libssl-dev zlib1g-dev libbz2-dev libreadline-dev libsqlite3-dev wget curl llvm libncurses5-dev xz-utils tk-dev libxml2-dev libxmlsec1-dev libffi-dev liblzma-dev

then update_requirements.sh failed with cannot install pycurl

ended up fixing by first running
sudo apt install libcurl4-gnutls-dev

pd0wm, 2021-06-07:
Did you run `ubuntu_setup.sh`? That sounds like the place where this package should installed. That line only installs the dependencies needed to install pyenv itself. Packages needed for openpilot dependencies live in `ubuntu_setup.sh`, like `libusb-1.0-0-dev` which is needed for the `libusb1` pip package for example.

ClockeNessMnstr, 2021-06-07:
I did, I was fumbling through the install. I was working into the tool directory so I ran update_requirements then Ubuntu_setup and then install.sh for plot juggler

pd0wm, 2021-06-07:
In the dockerfile and the ubuntu setup script we install `libcurl4-openssl-dev`, which seems to work fine in CI. Any reason your pycurl want the gnutls version instead of the openssl one?

ClockeNessMnstr, 2021-06-07:
Not that I can recall from any failed messages.

ClockeNessMnstr, 2021-06-07:
I probably botched it somehow

closed 2021-06-08 by adeebshihadeh