commaai/openpilot #21162: Model failes to properly handle "fifth lane" turn lanes.

nabeelr, 2021-06-06:
**Describe the bug**

**How to reproduce or log data**

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

**Additional context**

HaraldSchafer, 2021-09-04:
Closing this, this is part of the general improving laneless driving project. Not really actionable with a specific fix, thanks for the report!

closed 2021-09-04 by HaraldSchafer

nabeelr, 2021-09-05:
Thanks for the input Harald. 

Should I avoid these kind of reports in the future?

Any feedback on making my feedback more valuable to the team?

HaraldSchafer, 2021-09-06:
For us they can still be useful, especially detailed ones like this. It keeps us aware of the failures people are experiencing and gives us data to look at to test fixes for certain failures. 

But we do prefer to have them closed if there is not an immediate fix in sight.

nabeelr, 2021-09-07:
Cool, good to know. If I'm every submitting things that aren't helpful or too often, please let me know. 😉