commaai/openpilot #21163: report the error "Immediately connect to the internet to check for updates" when input "PASSIVE=0 NOSENSOR=1 USE_WEBCAM=1 ./manager.py

LIHaoran-good, 2021-06-06:
**Describe the bug**
I use the webcam tool to run openpilot on PC. We refer to the [guideline](https://github.com/commaai/openpilot/tree/master/tools/webcam) and connected the PC to two web cameras and Panda. When input "PASSIVE=0 NOSENSOR=1 USE_WEBCAM=1 ./manager.py", report "Immediately connect to the internet to check for updates",  details please refer to below picture. How to solve it ? Thanks.
[image]

**Expected behavior**

Openpilot can run successfully

**Additional context**

Operating system: [e.g. Ubuntu 【20.04]

pd0wm, 2021-06-07:
Have you tried running updated on PC? This can be solved by clearing your params folder (`~/.comma/params`)

adeebshihadeh, 2021-06-07:
Fixed in 29ac94b71997303fdd2cae137a1a3ccd016868e7. After clearing your params, you shouldn't see this again.

closed 2021-06-07 by adeebshihadeh