commaai/openpilot #21164: VIDEOIO ERROR: V4L2: Could not obtain specifics of capture window. VIDEOIO ERROR: V4L: can't open camera by index 1

LIHaoran-good, 2021-06-06:
**Describe the bug**
I use the webcam tool to run openpilot on PC. We refer to the [guideline](https://github.com/commaai/openpilot/tree/master/tools/webcam) and connected the PC to two web cameras and Panda. When input "PASSIVE=0 NOSENSOR=1 USE_WEBCAM=1 ./manager.py", report the error "VIDEOIO ERROR: V4L2: Could not obtain specifics of capture window. VIDEOIO ERROR: V4L: can't open camera by index 1", details please refer to below picture. How to solve it ? Thanks.
[image]

**How to reproduce or log data**

When input "PASSIVE=0 NOSENSOR=1 USE_WEBCAM=1 ./manager.py"

**Expected behavior**

**Additional context**

Operating system: [e.g. Ubuntu 20.04]

iHelp101, 2021-06-06:
Run "ls /dev/video*" to find your webcam IDs. It is likely that the webcams are not mapped to the standard IDs. You can then adjust camera_webcam to match the IDs.

LIHaoran-good, 2021-06-07:
@iHelp101 Thanks for your reply.   After I verified the webcam IDs in camera_webcam.cc,  it is ok

closed 2021-06-07 by pd0wm