commaai/openpilot #21170: On master-ci 0.8.5, device menu and buttons can be accessed while on-road and driving.

nabeelr, 2021-06-07:
**Describe the bug**
After a fresh boot, you can access the buttons in the device menu while OP is on road and engaged.
Accessing the preview of the driver monitoring camera will cause OP to disengage unexpectedly and for an LKAS fault to appear on screen in OpenPilot. After this, the items in the device menu will be greyed out, and inaccessible. You will be able to engage OP again after a few seconds.

**How to reproduce or log data**

Start with a C2 that is powered off. Let it boot up, and then start driving. Engage OP, click on the gear button, and select the driver monitoring preview button while OP is engaged. OP will disengage unexpectedly and display an LKAS error.

**Expected behavior**

The buttons in the device menu should always be inaccessible when on road.

**Device/Version information (please complete the following information):**
 - Device: C2
 - Dongle ID: 903210ecc0cb706b
 - Route: 903210ecc0cb706b|2021-06-06--23-21-01--0
 - Timestamp: 23:21:49
 - Version: master-ci openpilot-v0.8.5-4978a7fae-2021-06-07
 - Car make/model: 2019 Golf R

**Additional context**

None

nabeelr, 2021-06-08:
@adeebshihadeh I was actually able to reproduce it even if my C2 was booted with the ignition off. It seems to be a problem after first boot, until you trigger the problem, then you get the LKAS error, and after that, it will behave properly, with the buttons being disabled when the ignition is on and on-road.

adeebshihadeh, 2021-06-08:
Fixed in 1c952e75b58b1733742644a3d77bf57cc7d1818f

closed 2021-06-08 by adeebshihadeh