commaai/openpilot #21171: ui interface displays NO Panda

LIHaoran-good, 2021-06-07:
**Describe the bug**

I use the webcam tool to run openpilot on PC. We refer to the [guideline](https://github.com/commaai/openpilot/tree/master/tools/webcam) and connected the PC to two web cameras and Panda. When input "PASSIVE=0 NOSENSOR=1 USE_WEBCAM=1 ./manager.py", the webcam view can be displayed , but displays NO Panda, details please refer to below picture. How to solve it ? Thanks.

[image]

**How to reproduce or log data**
input "PASSIVE=0 NOSENSOR=1 USE_WEBCAM=1 ./manager.py"

**Expected behavior**

**Additional context**

Operating system: [e.g. Ubuntu {20.04]

pd0wm, 2021-06-07:
Can you post your manager.py output?

adeebshihadeh, 2021-06-07:
We like to keep our issue tracker to bugs and enhancements. Our [community discord](https://discord.comma.ai/) is a better place to get openpilot running.

closed 2021-06-07 by adeebshihadeh