commaai/openpilot #21193: Jittery acceleration when going downhill. OP will bounce on Max speed matching.

samabenie1, 2021-06-08:
**Describe the bug**
When setting a max speed and coasting downhill with Openpilot, openpilot seems to overcompensate for the coasting and decrease your speed too much. This results in bouncing between max speed and 1mph below max speed. It can be felt in the car and seen on the tachometer

**How to reproduce or log data**

enable autopilot at any max speed. Proceed towards a road decline. If the decline would increase your speed above the openpilot max speed, then the issue seems to occur every time. This did not happen with stock ACC.

log data made public

**Expected behavior**
open pilot will pick a speed for the decline and stay there. Rather than bouncing around. 

**Device/Version information (please complete the following information):**

    Device: C2
    Dongle ID: f080b0eff75e609f
    Route: f080b0eff75e609f/1623112435320/1623113870691
    Timestamp: f080b0eff75e609f|2021-06-07--19-33-55--16
    Version: 0.8.4
    Car make/model: 2019 Honda Odyssey

**Additional context**

chicman, 2021-06-12:
I would second this issue is annoying.

xps-genesis, 2021-07-11:
@pd0wm This cannot be optimally fixed without introducing road slope into op long calculation.

road slope = (long ax from intertial sensor) - (ax from wheel). with some filtering?

final_accel = (road slope + output_gb)

eg.
on a 10% slope downhill, or a ~1m/s2 effective acceleration due to slope, the braking should start form (output_gb < 1m/s2) and not (output_gb <0 m/s2) like it is set up to do now. the current setup will cause a inital high error(1m/s2 step) in the brake controller causing it to over brake and release and oscillate.

if we want to maintian a constant speed on a downhill slope, the braking accel to the brake controller should be equal to effective accel(g*sin(theta)) due to roadslope.

adeebshihadeh, 2021-09-04:
The Honda Odyssey tuning is being addressed in https://github.com/commaai/openpilot/issues/21697.

closed 2021-09-04 by adeebshihadeh