commaai/openpilot #21200: OpenPilot on Jeep Gr Cherokee Gives no audible signal when ceasing steering control

Trumanlhesterjr, 2021-06-08:
Need audible when I have to Take control

VirtuallyChris, 2021-06-08:
Can you please fill out the template so we can have the information necessary to take a closer look?

Trumanlhesterjr, 2021-06-09:
Thank you for your response. This is my first attempt
at issue reporting
Obviously I,m new to this process. I don't know where the
"template" is or the data that should be put in it.
I Have the following data
Car  2021 Jeep Grand Cherokee Overland
Version   openpilot V0.8.4 - release
Get Branch release 2
Get commit ea0e4bocbb
Panda Firmware  954c 472a 19be 2963
OS Version NEOS162
Dongle ID 6308 7bc1 e5d3 7e89
Serial 8bed cbdf

My system works perfectly as designed.
Starts Lane sense control and steering at
39 mph and continues at speeds above.
When speed drops to below 39 mph lanne
sense control and steering ceases.
There is a visual alert on the device screen when
this occurs to "Take Control"
However there is no audible alert.
If you are not paying atten to the device
for the visual alert, the first indication that you
have that the system has relinquished control
is when you are drifting into another lane
God an optimal situation

Thank you again for looking into this matter
Truman

closed 2021-06-17 by adeebshihadeh

Trumanlhesterjr, 2021-06-18:
> Closed #21200 <https://github.com/commaai/openpilot/issues/21200> via

Thanks for the quick action