commaai/openpilot #21201: 21 Hyundai Elantra Hybrid can error: check connections

sebastian4k, 2021-06-08:
Booting up with both release2-staging and master-ci branch to access the 21 Elantra support I get Can error: check connections. Not able to get past that to run openpilot.

Device/Version information:
 - Device: comma two
 - Dongle ID: d73f21a646ca211b
 - Route: d73f21a646ca211b|2021-06-08--15-34-04
 - Version: 0.8.5 release2-staging, 0.8.5 master-ci
 - Car make/model: 2021 Hyundai Elantra Hybrid Limited

VirtuallyChris, 2021-06-08:
The Elantra Hybrid isn't supported so that's likely the cause

adeebshihadeh, 2021-06-10:
Chris is correct that the Elantra Hybrid isn't supported. We'd be happy to help get your car supported and working if you open a PR, but this isn't an issue as new car support comes from the community. Check out our community discord if you need help.

closed 2021-06-10 by adeebshihadeh