commaai/openpilot #21202: Replace frame stream with replay

adeebshihadeh, 2021-06-09:
Now that we have a nice way to use vipc from python, we no longer need camera_frame_stream. We still need a default camera driver for PC, so I think it would be nice to replace it with a replay using one of the public [test routes](https://github.com/commaai/openpilot/blob/master/selfdrive/test/test_routes.py).

adeebshihadeh, 2021-06-12:
@deanlee interested in this one?

deanlee, 2021-06-12:
do we need to support driver camera in camera_pc? there is no dcamera in public test ruotes

adeebshihadeh, 2021-06-12:
I’d like to support it. Here’s a public route with dcam: 534ccd8a0950a00c|2021-06-08--12-15-37.

deanlee, 2021-06-12:
https://commadataci.blob.core.windows.net/openpilotci/534ccd8a0950a00c/2021-06-08--12-15-37/0/dcamera.hevc:
BlobNotFoundThe specified blob does not exist

deanlee, 2021-06-12:
#21241

closed 2021-06-18 by adeebshihadeh