commaai/openpilot #21239: Complete shutdown while updating on freeway

fadapter, 2021-06-12:
Open pilot froze in the middle of the freeway I think it was updating, my wife was driving she got freaked out, really close call, car shut down completely could not restart  till a few minutes, 2020 keep gran wagoner summit. 
Super fan of open pilot.
Felipe9595@yahoo.com

adeebshihadeh, 2021-06-12:
Can you fill out the bug report template so we can look into this? The updater only runs while the car is off; why do you think it was updating?

fadapter, 2021-06-12:
I am not Sure how to fill a bug report, the car shut down that’s when I saw that it was updating, we wore driving on I 10 west bound to Tucson from El Paso Tx, there was very bad internet.

adeebshihadeh, 2021-06-12:
The template is given when you open a bug report. Here it is:
**Describe the bug**

**How to reproduce or log data**

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

**Additional context**

adeebshihadeh, 2021-06-14:
Closing due to inactivity. Feel free to open a new issue with the template filled out

closed 2021-06-14 by adeebshihadeh