commaai/openpilot #21246: 0.8.5 Breaks Honda Accord Support (2018-2020): CAN Error: Check Connections

ChadBailey, 2021-06-12:
**Describe the bug**

With no other changes, updating from 0.8.4 to 0.8.5 made the 2018 Honda Accord stop working - me and others are reporting the same (see discord #honda-acura channel). C2 now displays "CAN Error: Check Connections"

**How to reproduce or log data**

Update to 0.8.5 using a Comma 2 and 2018 Honda Accord

**Expected behavior**

CAN bus continues to be operational allowing openpilot to drive the vehicle :P

**Device/Version information (please complete the following information):**
 - Device: comma 2
 - Dongle ID: 053abdc1032b7ecf
 - Route: N/A (but this is a recorded route generated while I was attempting to troubleshoot) 053abdc1032b7ecf|2021-06-12--14-08-37
 - Timestamp: N/A
 - Version: v0.8.5-release [branch release2, commit ae7ac2242d]
 - Car make/model: 2018 Honda Accord (2020 Honda Accord has been reported as also impacted, guessing impacts all bosch models)

**Additional context**

I reseated all connections and rebooted the device/car multiple times to be doubly-sure there was nothing more I could do to resolve it, it had no impact.

ChadBailey, 2021-06-12:
Based upon the nature of the changes, I suspect this may be related to the implemented fix for https://github.com/commaai/openpilot/issues/20872

sushantshah1991, 2021-06-12:
I can provide a 2020 Honda accord dongle id of required. Let me know...

ShaneSmiskol, 2021-06-12:
Make sure upload raw logs is enabled and that the device is on WiFi so that the logs can upload

ChadBailey, 2021-06-12:
Currently have my c2 unplugged so that I could use the stock ACC. I'll go do that here in the next 30 mins or so

adeebshihadeh, 2021-06-12:
This is likely a regression from #21105. This should be fixed with #21247.

ChadBailey, 2021-06-12:
Thanks for the quick turnaround! Now that the issue is understood I've updated the title to be more precise.

adeebshihadeh, 2021-06-13:
Thanks for reporting this. Fixed in af8d4176271c0ecc972d1e8d437aa1707abaf5d1.

closed 2021-06-13 by adeebshihadeh

ChadBailey, 2021-06-13:
For the benefit of any followers, I can confirm this is now fixed. Thanks again!

sushantshah1991, 2021-06-13:
Okay, so will the update be automatically pushed to the device or need to uninstall and reinstall the software?

ChadBailey, 2021-06-13:
For me, I did a manual check for updates and it prompted to update again. It applied the update and rebooted as usual. It still said 0.8.5 but began working.

dontcallmereggie, 2021-06-16:
I'm confused as to what to do. 2019 Honda Accord 2.0.