commaai/openpilot #21268: Meta: Community Wiki Changes Immediate

ChadBailey, 2021-06-15:
I recently modified the community wiki page https://github.com/commaai/openpilot/wiki/SSH and the changes were live immediately with no approval process (see the [history](https://github.com/commaai/openpilot/wiki/SSH/_history) for my change). I'm not even certain anyone is notified that changes have been made. Can I have someone peer review the changes I've made? (I added the Experienced Developer Quick Start)

This behavior may be intentional, but I looked for anything discussing this and couldn't find it. I'm opening this report since the policy could be abused by spam bots.

Best of memory, when I've made similar changes in other projects it generated a pull request to the wiki directly in browser. Perhaps my mind has fabricated this memory though.

adeebshihadeh, 2021-06-15:
This is intentional. The README's in the repo are intended to be authoritative, while the wiki is community maintained (see the [SSH README](https://github.com/commaai/openpilot/blob/master/tools/ssh/README.md) as an example). AFAIK, there is no feature like the one you remember in GitHub Wikis.

closed 2021-06-15 by adeebshihadeh

ChadBailey, 2021-06-15:
Thanks for the explanation. This open-and-shut bug report should now help future people curious about this, so I think it was worth the effort in the end.