commaai/openpilot #21284: OP failed to brake when lead car moved little bit

adnanlanewla, 2021-06-15:
**Describe the bug**
OP was following the lead car properly. However, when the lead car creeped ahead little bit OP accelerated abruptly and failed to brake for the lead car.

OP was following the lead properly. However, as soon as the lead car creeped a bit, OP accelerated too abruptly resulting me in disengaging OP

**How to reproduce or log data**
N/A

N/A
**Expected behavior**
OP should start not accelerate too much when lead car creeps a bit

I also expect that OP should be braking sooner when following the lead car

**Device/Version information (please complete the following information):** 
 - Device: [e.g. EON/EON Gold/comma two] (comma2)
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]  (d2ef5aa690e849d0)
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin] [d2ef5aa690e849d0|2021-06-14--22-35-13--4]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)] (7s-14s)
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)] (0.8.5)
 - Car make/model: [e.g. Toyota Prius 2016]  (2020 Toyota Rav4)

**Additional context**
Here's the youtube video of the segment. Timestamp (7-14secs)
https://youtu.be/vOOYIaejUoQ

pd0wm, 2021-06-16:
[image]

Duplicate of https://github.com/commaai/openpilot/issues/2731

closed 2021-06-16 by pd0wm

adnanlanewla, 2021-06-16:
Any idea why the car is slow to respond to the braking command ? This is my second bug of the same issue and others with TSS2.0 have experienced the same issue. 
If we don't know the reason for the car slow in responding to the braking command. Then the alternate way to mitigate the issue  is to not accelerate abruptly when the lead car creeps ahead. Would that work ?

pd0wm, 2021-06-16:
We’re working on a complete overhaul of longitudinal control, so we are not making tweaks like that to the current implementation. Once the rewrite is done we’ll look more closely at these car specific quirks.