commaai/openpilot #21285: OP accelerated while the lead car was stopped

adnanlanewla, 2021-06-15:
**Describe the bug**
OP accelerated while the lead car was stopped at a red light

The lead car in front was stopped and the OP braked properly. However, after sometime it just accelerated for no reason resulting me manually taking over

**How to reproduce or log data**
N/A

N/A
**Expected behavior**
OP shouldn't accelerate when the lead car in front is stopped

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two] (comma2)
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin] (d2ef5aa690e849d0)
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin] [d2ef5aa690e849d0|2021-06-13--18-29-03--23]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)] (32-35secs)
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)] (0.8.5)
 - Car make/model: [e.g. Toyota Prius 2016] (2020 Toyota Rav4)

**Additional context**
Here's the youtube video of the segment. Timestamp 32-35 secs
https://youtu.be/TZniEcVTrbQ

pd0wm, 2021-06-16:
Thanks for this report. Looks like the lead car velocity from the radar indicated the car moved forward. Looks like something weird was going on with the detection of the lead car in general, since in the few seconds before this happened it actually thought the lead car was moving backwards.
[image]

adnanlanewla, 2021-06-16:
As you can see from the video that the lead car was still between 32-36secs. And As a driver too I never noticed the lead car moving even a tiny bit.

pd0wm, 2021-06-16:
The frame rate of your posted video is slightly off. A segment should be 60 seconds, not 48. So this disengagement at 42 seconds aligns with 34 seconds into your video.

adnanlanewla, 2021-06-16:
That's true not sure why the frame rate is off. The disengagement does align with 34 secs in my video

HaraldSchafer, 2021-06-16:
The model predicted the lead was moving forward at 1mph, so the model messed up. 

The car also got way closer to the lead before that than it should have, probably the downhill? This puts it in a very close situation which makes the model confused because it is so zoomed in. 

Multiple things seem to have compounded in this case. We are currently working on longitudinal quite a bit and making test cases for this in simulation, as well as working on giving the model a bigger FOV.

adnanlanewla, 2021-06-17:
Hope to see those long changes soon.

bradmurray, 2021-07-23:
I'd like to piggyback on this report as I have started seeing the same thing recently. I've been on 0.8.5 for a while now. Recently I can't trust the thing to come to a stop behind a stopped car. The triangle on the HUD disappears when I get close to the lead car while still rolling, the path line gets all squirrely and then the car starts accelerating. There is enough room and time to react, but it definitely should not be doing this and I never noticed it happening until recently.

Device: comma two (w/ smartDSU)
Dongle ID: a2b1d650214053d7
Car make/model: Toyota Prius Prime 2020
Version: 0.8.5

Route: a2b1d650214053d7|2021-07-22--20-02-11
Timestamps: 20:42:54, 20:44:27, 20:53:30, 20:55:40

Route: a2b1d650214053d7|2021-07-22--07-57-00
Timestamps: 08:04:53, 08:23:22

Route: a2b1d650214053d7|2021-07-21--08-42-01
Timestamps: 08:44:05

adnanlanewla, 2021-07-23:
You should try 0.8.6. It's much more stable longitudinal  wise

HaraldSchafer, 2021-07-23:
Yeah it should be smoother.

closed 2021-10-04 by pd0wm

Sparra215, 2022-01-19:
My car was stopped in traffic, the lead car was stopped but my car rolled forward and hit the other vehicle. I did not have time to react. Would you mind helping me figure out why this happened? Thank you. 

Device: Comma 3
Dongle ID: d250f6ef680e703d
Car: 2021 Chrysler Pacifica Hybrid
Version: 0.8.12

Route: d250f6ef680e703d\|2022-01-15--12-13-43
--

Time stamp: 2022-01-15--12-13-43--44

HaraldSchafer, 2022-01-19:
@Sparra215 You're using a fork that does god knows what to longitudinal control. Either way, openpilot did not request acceleration.

You mention no time to take over, but there is over 3seconds between your car beginning to creep and the collision.

Sparra215, 2022-01-19:
Thank you for looking into it. No I didn’t think OP requested the
acceleration I know my vehicle uses stock ACC. Yeah looking back at the
video it was a slow creep, I just didn’t realize it. I don’t blame the
technology, I know I was in control and should have paid better attention.

I was just hoping to find out what the stock ACC was telling the car at
that moment. Or what the orange colored alert was during the roll forward.

Thank you for your help.

Alex

Sparra215, 2022-01-19:
In the video segment of my car rolling forward, there is a orange segment
of video. Is there any way to find out what message OP was sending me with
that orange alert at the time the car rolled forward?