commaai/openpilot #21287: Hugging Center Line, Changing between 2019 Camry and 2019 RDX

MikeFoerster, 2021-06-16:
I purchased the Comma 2 less than 3 months ago.  I started running it in my 2019 Camry, which worked fine.
      I was told by support at Comma.ai, that I could move the OP between 2 vehicles. 

My son added the signature for the 2019 RDX.  
  (brtk on discord, he is a hell of a lot better programmer than I am, I was a software test engineer in a past life)  
   The 2019 RDX is now supported in the  v0.8.4-release, which I recently loaded.

The problem basically is that when I started driving the Camry down the road today (after moving the device from the RDX), the car will hug the center line, almost on top of it.   In the "Line" mode, the lines that should indicate what the OP is finding for the outside lines of the lanes, are no longer indicating the road edges, but rather track inside the road lines in a closing inverted "V" formation.

I "suspect" that this may have been caused when I moved the OP (Comma 2) from the 2019 Camry SE to the 2019 RDX Tech Pkg.

This same problems has occurred twice now.  The first time when I was running the GIT branch that my son created.  The second time was today.  My wife had driven the RDX on a 450 mile trip this last weekend and returned.  I then moved the OP back to the Camry this morning and discovered that the problem had in fact, occurred for the 2nd time.   

The first time this happened was on about 6/3/21 but I was running the GIT branch that my son created.  I consequently reloaded the v0.8.4-release code.

I tried contacting support@comma.ai after the first incident but they only asked questions, and suggested that I enter a bug report.

I'm not certain yet on exactly how to reproduce the behavior, other than the swap the OP between the two cars a few times.  

I'm hoping that someone can explain this erratic behavior 
   I do have a couple pictures that I took with my phone showing the "V" lines in the road ahead.

**Device/Version information (please complete the following information):**
 - Device: Comma 2
 - Dongle ID: 6dce577e927e86b3
 - Route: I can't get the route number because with my OP here in the house, my OP appears Off-Line
 -         I don't understand why there are no drives showing!!!!
 - Timestamp: I don't have the exact "timestamp" but it was at about 10:00 AM 6/16/21.
 - Version: v0.8.4-release
 - 2019 Toyota Camry (was removed from the 2019 Acura RDX

**Additional context**

please contact me at **mcfroston@gmail.com**

[image]

adeebshihadeh, 2021-06-16:
I checked your device's logs, and it seems calibration wasn't reset when the device was moved. The behavior you're seeing is a result of the device starting with invalid calibration values. Let us know if you still have issues after resetting calibration in the settings.

closed 2021-06-16 by adeebshihadeh

MikeFoerster, 2021-06-19:
Am I supposed to reset the calibration each time I move it between vehicles?
   I was not aware of that!!!  Nothing was said when I inquired about moving it between vehicles.

Mike

adeebshihadeh, 2021-06-19:
Yes, reset each time the device is moved.