commaai/openpilot #21307: Stuck on "Registering"

sethprice240, 2021-06-17:
**Describe the bug**

Idk

I installed the shane thing. 
Stuck on registering. 

Installed the factory. It worked.

Tried to install the shane again thru diff link. Stuck on registering 

Tried to install factory again, stuck on registering

**How to reproduce or log data**

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: [comma two]
 - Dongle ID: [70cc8712c411de8b
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [0.8.5???
 - Car make/model: [2021 corolla hybrid 

**Additional context**

adeebshihadeh, 2021-06-17:
Does your device have an internet connection while it's trying to register?

sethprice240, 2021-06-17:
Yes. But is/was tethered to a my cellphone while riding in the car and updating. So maybe not the MOST stable

adeebshihadeh, 2021-06-17:
Your description wasn't clear, but I'd recommend factory resetting your device and installing stock openpilot. Make sure to stay within range of a stable wifi network while installing until registration is done. If there are issues with a fork, contact the fork maintainer.

sethprice240, 2021-06-17:
The issue was the internet connection.

Thank you

closed 2021-06-17 by adeebshihadeh