commaai/openpilot #21323: Connect to the Internet to check for updates

rumym, 2021-06-18:
Hi, we are currently in Canada and I keep getting the below message on my comma ai. I connected to my mobile hotspot and then it lets me drive however, once I restart the car and it is no longer connected to my mobile hotspot it presents the same message again.

Immediately connect to the Internet to check for updates. If you do not connect to the Internet open pilot want to engage in 0 days.

adeebshihadeh, 2021-06-18:
The updater only runs while the car is off, so you'll need to keep it off and connected to internet until you see the update prompt. We allow the device to go onroad the first time after boot, which is why you it works sometimes for you. The latest release has some updater controls which should help your situation next time.

closed 2021-06-18 by adeebshihadeh

rumym, 2021-06-18:
I have connected it while the car is off as well as ACC mode but still nothing happens.

adeebshihadeh, 2021-06-18:
It's downloading a large update. You can uninstall and reinstall openpilot if you want to watch the progress.

rumym, 2021-06-18:
Do I need to do this in the car or can I bring the phone unit into the house and do the update while it is disconnected from the car?