commaai/openpilot #21333: hyundai: inconsistent fingerprinting

gregjhogan, 2021-06-18:
sonata and palisade (perhaps others?) will sometimes fuzzy fingerprint even though no ecus have had a firmware update

example route: 24804955a125718c|2021-06-13--15-31-24

I see what happened is we query a few data ids and it does not get a response for data id \xf1\xa0 (normally I guess it does).  It looks like there are many cases where we have the same ecu fw versions for a vehicle with and without \xf1\xa0  at the end, and other ecus never return a value for this data id.  I also see the data returned by this id is pretty much always included in another id that is being queried.

Since you cannot run a script and capture all combinations easily we could
* drop \xf1\xa0 id from the query list
* retry fingerprinting a few times

adeebshihadeh, 2021-06-19:
#19887 is similar

Jon8RFC, 2021-06-29:
Since 0.8.5, my 2018 Hyundai Sonata has often displayed the "no exact match" warning.  Is it due to fuzzy and/or because of the [camera fw removal from sonata_lf](https://github.com/commaai/openpilot/pull/20850/commits/f28f966c3447f1105b20c376812711f3631632e3#diff-211ba5315f2af5d2eb34a89b5ac67ade2db803c4f61a92150a0541aa41e19f51)?
This firmware is one of those inconsistent ones which rarely shows up, but here it is:  6363a0e48f515f27|2021-06-04--17-09-42--0

sunnyhaibin, 2021-09-25:
This issue also affects my 2021 Hyundai Sonata N Line. Here’s the route ID for one of the drives: `36e10531feea61a4|2021-09-22--17-51-10`

sunnyhaibin, 2021-09-25:
2021 Kia K5 affected by the same issue: `007d5e4ad9f86d13|2021-09-25--17-38-12`

pd0wm, 2021-09-25:
Just merged some improvements (https://github.com/commaai/openpilot/pull/22344/files). Let's see if that improves things.

sunnyhaibin, 2021-09-26:
> Just merged some improvements (https://github.com/commaai/openpilot/pull/22344/files). Let's see if that improves things.

Tried it already, the route IDs provided earlier were using those improvements, the issue still comes in frequently. Sometimes it would take about 5 times to have the car recognized.

sunnyhaibin, 2021-09-27:
The latest PR of the improvements seen to have made the matter worse... I had to turn on/off for almost 15 times in 1 instance for my car to get recognized.

Car: 2021 Hyundai Sonata N Line
Version: `master-ci` as of 9/25/2021

[image]

pd0wm, 2021-09-27:
@sunnyhaibin Can you make sure rlogs for the K5 and Sonata drives are uploaded so I can take a look at the CAN traffic?

sunnyhaibin, 2021-09-27:
@pd0wm Just checked in useradmin, my Sonata drives should be all uploaded now. K5 is unconfirmed but should also have the drives uploaded.

sunnyhaibin, 2021-09-30:
@pd0wm The issue is still relevant with recent commits: 
[image]

pd0wm, 2021-10-06:
@sunnyhaibin I did some digging into your issue and found that messages from your transmission seem to be dropping. The first byte of the messages is a counter. In the first example `23` is missing, and in the second one it's `25` (see the iso-tp spec for more info). This throws an assertion in the uds.py code. This exception is only caught in an outer loop causing quite a few results to be lost.

@gregjhogan Do you have any idea what could be causing these messages to disappear? Is the ECU just buggy or is it lost somewhere on our side.

Route `36e10531feea61a4|2021-09-26--21-25-19`
```
t, bus, addr, data, ascii
61761833311 	 129 	 0x000007e1 	 b'0322f10000000000' 	 "� 
61772843572 	 1 	 0x000007e9 	 b'102b62f100543032' 	 +b�T02 
61782007634 	 129 	 0x000007e1 	 b'3000000000000000' 	 0 
61782007634 	 1 	 0x000007e9 	 b'21363031424c2020' 	 !601BL   
61782007634 	 1 	 0x000007e9 	 b'2254303237333041' 	 "T02730A 
61782007634 	 1 	 0x000007e9 	 b'2454323558585837' 	 $T25XXX7 
61782007634 	 1 	 0x000007e9 	 b'2533304e5335f75f' 	 %30NS5�_ 
61782007634 	 1 	 0x000007e9 	 b'2692f5aaaaaaaaaa' 	 &������� 
FW query exception: Traceback (most recent call last):
  File "/data/openpilot/selfdrive/car/fw_versions.py", line 282, in get_fw_versions
    fw_versions.update(query.get_data(t))
  File "/data/openpilot/selfdrive/car/isotp_parallel_query.py", line 111, in get_data
    dat: Optional[bytes] = msg.recv()
  File "/data/openpilot/panda/python/uds.py", line 401, in recv
    self._isotp_rx_next(msg)
  File "/data/openpilot/panda/python/uds.py", line 443, in _isotp_rx_next
    assert self.rx_idx & 0xF == rx_data[0] & 0xF, "isotp - rx: invalid consecutive frame index"
AssertionError: isotp - rx: invalid consecutive frame index
```

```
54152195241 	 129 	 0x000007e1 	 b'0322f10000000000' 	 "� 
54172577012 	 1 	 0x000007e9 	 b'102b62f100543032' 	 +b�T02 
54182268106 	 129 	 0x000007e1 	 b'3000000000000000' 	 0 
54182268106 	 1 	 0x000007e9 	 b'21363031424c2020' 	 !601BL   
54182268106 	 1 	 0x000007e9 	 b'2254303237333041' 	 "T02730A 
54182268106 	 1 	 0x000007e9 	 b'2331202056444e38' 	 #1  VDN8 
54182268106 	 1 	 0x000007e9 	 b'2454323558585837' 	 $T25XXX7 
54182268106 	 1 	 0x000007e9 	 b'2692f5aaaaaaaaaa' 	 &������� 
FW query exception: Traceback (most recent call last):
  File "/data/openpilot/selfdrive/car/fw_versions.py", line 282, in get_fw_versions
    fw_versions.update(query.get_data(t))
  File "/data/openpilot/selfdrive/car/isotp_parallel_query.py", line 111, in get_data
    dat: Optional[bytes] = msg.recv()
  File "/data/openpilot/panda/python/uds.py", line 401, in recv
    self._isotp_rx_next(msg)
  File "/data/openpilot/panda/python/uds.py", line 443, in _isotp_rx_next
    assert self.rx_idx & 0xF == rx_data[0] & 0xF, "isotp - rx: invalid consecutive frame index"
AssertionError: isotp - rx: invalid consecutive frame index
```

pd0wm, 2021-10-07:
I've also seen it happen on the camera, but not on other ECUs.

adeebshihadeh, 2021-12-15:
Is this still an issue?

Jon8RFC, 2021-12-15:
I did have "no exact match" on 0.8.10, but didn't drive enough on 0.8.11+ to give it a full trial.  It's infrequent for me.

Jon8RFC, 2022-04-09:
@adeebshihadeh I haven't had "no exact match" in 2022 on the Comma3.

celluj34, 2022-06-23:
Perhaps I'm wildly out of scope here, but why check multiple times? If you get a valid ECU, then you can stop checking? Surely it can't change unless you move the unit to a different car.

I know that wouldn't _fix_ anything, I guess I'm just curious.

closed 2023-02-01 by adeebshihadeh