commaai/openpilot #21344: [DM] Expand Area Used For Facial Tracking in Driver-Facing Camera

BadOlives, 2021-06-19:
My device is mounted in a position where the driver-facing cameras can still see my face, but only half of my face fits in the frame currently used for Driver Monitoring. This is causing DM to think that a driver is not looking at the road, when indeed I am. The current size of the frame being used restricts the mounting position of the device. I am suggesting to expand the frame to cover that gap not used for DM currently or too increase the width closer to the left edge of the screen (if looking at the preview in Driver Camera) in left-hand drive mode. Also incorporate this into right-hand drive mode by doing the equivalent to the right edge. This will allow users to have a bit more freedom in mounting their devices without having to worry about OP not working properly due to DM issues.

ZwX1616, 2021-06-21:
I believe we are always using method#3
https://github.com/commaai/openpilot/blob/28694041d5ee860cdd5f26acb63887e0ce7d8bbf/selfdrive/modeld/models/dmonitoring.cc#L71

I suggest to try aiming it a little bit yawed while staying in calibration limits

closed 2021-06-21 by ZwX1616

BadOlives, 2021-06-21:
> I believe we are always using method#3
> https://github.com/commaai/openpilot/blob/28694041d5ee860cdd5f26acb63887e0ce7d8bbf/selfdrive/modeld/models/dmonitoring.cc#L71
> 
> I suggest to try aiming it a little bit yawed while staying in calibration limits

I think the reason for that gap is because that is where the side UI usually is (if enabled to display with OP active). Possibly have the frame ignore the side UI?