commaai/openpilot #21346: Model (desiredaAngledDeg) oversteering.

ClockeNessMnstr, 2021-06-20:
On sharper turns the model is making sharp corrections after oversteering long turns.

With the Hud you can especially see the sharp changes in target directory as the model first oversteers and then tries to correct. 

Route segment:
a3586311666ae480|2021-06-18--16-44-45--10

In the PID log we can see the target angle is achieved well and not overshot. 

First, the oversteer, it noticably moves to the inside. This could be a slightly missed steering ratio being learned. Or maybe the model is cutting the corner.

Second, as the vehicle shifts in the lane. Their should be a shift in the desired angle. Instead it holds the high output almost all the way to the curb. Then DROPS the desired angle to correct. 

This happens every time.

ClockeNessMnstr, 2021-06-23:
[image]

ClockeNessMnstr, 2021-06-23:
going to try and figure out the tuning holding the angle for about 1 sec.

pd0wm, 2021-06-24:
As far as I can tell the turn is too tight to be fully visible in the limited model box. This causes the model to guess and overestimate the turn. When it gets farther into the turn it sees more and corrects.

ClockeNessMnstr, 2021-06-24:
makes more sense. when i get a chance, hopefully I can get a little more aggressive on the tune without making any oscillations anywhere else.

pd0wm, 2021-06-24:
You don't want to fix this with (PID) tuning. The only task of the PID controller is to follow the desired angle as closely as possible. If the model is wrong there isn't much you can do about it.

ClockeNessMnstr, 2021-06-24:
Thanks, you're correct. It won't be quite right/corrected. the PID looks just a touch sloppy tracking at the higher angle. looks like the desired angle is at least a little further from the lanes/curb even though it will still move around in the lane here.

ClockeNessMnstr, 2021-06-26:
[image]

ClockeNessMnstr, 2021-06-26:
This is on my other vehicle. the height should be the only difference. (both mounted centered and within 1deg up/lat)

Possibly not the model? unless it makes sense that height could interfere?

pd0wm, 2021-06-29:
In your example route I plotted the model on top of a map, and it was definitely overestimating the curvature of the turn.

closed 2021-06-29 by pd0wm

ClockeNessMnstr, 2021-07-04:
[image]

ClockeNessMnstr, 2021-07-04:
got better tracking on the forester. the ramp up is limited by the eps to about 30/2047 per 0.02s

pd0wm, 2021-07-05:
You're saying it was ignoring the request because the request was ramping up too quickly?

ClockeNessMnstr, 2021-07-05:
Not totally ignoring, but spending a lot of time in a steering "deadzone" and eps not keeping up with requested ramp up/down rate

 the eps wass actually decreasing torque longer than the LKAS was decreasing since even when lkas increases for a few steps it's lower than the eps torque reached. 

Some of the improvement is just setting the ramp_up limits even lower to match eps behavior. Then when the controller changes direction from the steering moving, the eps also changes immediately.

The rest of my improvement is the back calculation PR. Otherwise the integrator is winding up more against the ramp up limits.

ClockeNessMnstr, 2021-07-05:
[image]
[image]

ClockeNessMnstr, 2021-07-05:
different/exaggerated scenario but here is a recovery from holding the steering off course and releasing. Same tune but the first is unstable because of the lag in the torque request from the LKAS ramp up/down being to high.

pd0wm, 2021-07-05:
I agree that we should probably limit the integrator when it's hitting the ramp limits.

Have you tried using the `freeze_integrator` input to the pid controller? Maybe it's as simple as

`latcontrol_pid.py:42`
```python
      output_steer = self.pid.update(angle_steers_des, CS.steeringAngleDeg, check_saturation=check_saturation, override=CS.steeringPressed,
                                     feedforward=steer_feedforward, speed=CS.vEgo, deadzone=deadzone, freeze_intergrator=CS.steeringRateLimited)
```

ClockeNessMnstr, 2021-07-05:
freeze integrator if reaching the ramp up limits would work a little but we'd lose too much integrator action hitting the limits all the time. 
could also feed back a "saturation direction" (-1, 0, 1) to enable clamping on ramp up saturation like currently done with the absolute limits.
but, yes, at that point it's cleaner to just feedback the saturated output and we can back-calculating to get the most anti-windup benefit. 

In the screenshots above the integrator is actually reversed by the back-calculation during driver-override. This gives a smoother / more stable release of the steering wheel than if it was clamped at it's previous value.