commaai/openpilot #21349: DM false positives a lot at night (master-ci 086)

cydia2020, 2021-06-20:
**Describe the bug**

DM false positives a lot while driving at night under specific condition(s).

**How to reproduce or log data**

Use DM at night while slightly tilting head upwards. (This is my normal driving posture and it doesn't cause any issues during daylight time)

**Expected behavior**

DM should only alert driver when driver is actually distracted.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 399b1cca765e80b4
 - Route: I don't have uploader so unfortunately I can't provide one
 - Timestamp: as above
 - Version: 0.8.6 (master-ci)
 - Car make/model: Toyota Prius 2016

**Additional context**

adeebshihadeh, 2021-06-21:
We'll need a route to look into this. Can you enable the uploader and post a route?

adeebshihadeh, 2021-06-21:
I see you're running and maintaining a fork called "dodgypilot" which disables the uploader and claims to violate safety in ways that will get you banned. I suggest you revert the safety violations. If you have any issues on stock openpilot, feel free to open an issue, but we can't look into this without a route.

closed 2021-06-21 by adeebshihadeh