commaai/openpilot #21353: new EONS stuck at homepage?How to setup openpilot rightly?Dose this device wrok rightl?

liyangsensetime, 2021-06-20:
I follow the instruction (https://github.com/commaai/openpilot/wiki/Installing-openpilot),but my EONS just stuck at thiepage.
I tried to connect it to my car and the icon "nopanda" has been change into "conneted online".
[image]

liyangsensetime, 2021-06-20:
The version is 0.8.5.
i have cost more than 2 days to solve this problem ,i print some log in  this sdk showing that the road camera do not get connected. 
The driver camera could work rightly and i can get images of the road camera from comma app .
If someone can solve thie problem , i would appreciate it

cydia2020, 2021-06-21:
it seems like the device doesn't go "onroad", you should try another usb-c cable

liyangsensetime, 2021-06-21:
> it seems like the device doesn't go "onroad", you should try another usb-c cable
i have tried this way ,but this device just stuck  at this page ?
[image]

pd0wm, 2021-06-21:
It looks like you're not using the original USB-C cable. A lot of third party USB-C cables don't have all the pins connected internally. Make sure it supports USB 3.2 gen 2.

cydia2020, 2021-06-21:
這是甚麼車型？沒看到ADAS相機好像

What car model is this? It doesn't look like it has an ADAS camera.

pd0wm, 2021-06-21:
GitHub issues are for tracking issues related to openpilot. This issue is specific to your hardware or installation, and doesn't belong on GitHub. If there is an issue with your C2 or hardware bought from comma please contact support at support@comma.ai. For other questions join our discord on https://discord.comma.ai.

closed 2021-06-21 by pd0wm