commaai/openpilot #21355: Since v0.8.5, OP no longer keeps breaking when cruise disengages on Honda

jedberg, 2021-06-20:
**Describe the bug**

Prior to v0.8.5, if OP was breaking when cruise disengages, it would continue to brake behind the lead car all the way to 0mph.  Since v.0.8.5, it doesn't do that anymore.  It just disengages cruise and lets go of the brake.  The first time it happened it caught me off-guard because I was on a downhill and it sped up towards the car in front of me.

**How to reproduce or log data**

Get behind someone who slows from >22mph to <22mph while OP is engaged.

**Expected behavior**

I would expect the pre-0.8.4 behavior of continuing to break down to 0mph.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 0c769fcdc1652c86
 - Route: 
 - Timestamp: 
 - Version: 0.8.5
 - Car make/model: 2018 Honda Odyssey 

**Additional context**

I'm new to OP and filing issues, so please let me know if I did anything wrong.  I don't have Route and Timestamp info because I didn't think about filing an issue until just now, so I didn't note it down. :). I'll update with info if it happens again.

pd0wm, 2021-06-21:
Do you also have a route with the previous behavior?

jedberg, 2021-06-21:
@pd0wm Yes!  I managed to find one.

Route: 0c769fcdc1652c86|2021-05-29--18-07-40
Timestamp: 9 min 40 seconds into drive (18:17:20)

pd0wm, 2021-06-21:
Unfortunately the rlogs from that route are already deleted from our servers. Do you have anything from the last two weeks?

jedberg, 2021-06-21:
How about this one:

Route: 0c769fcdc1652c86|2021-06-07--16-10-02
Timestamp: Starts around 10 minutes and 10 seconds in (16:20:18)

jedberg, 2021-06-21:
@pd0wm Here's an even better one, all the way down to 0:

Route: 0c769fcdc1652c86|2021-06-09--11-14-49
Timestamp: Starts around 10 minutes and 30 seconds in (11:25:21)

sshane, 2021-06-24:
0c769fcdc1652c86|2021-06-07--16-10-02 is from 0.8.4 but has the same behavior you described in 0.8.5. The code in honda/interface.py has been there for over a year so you should have been experiencing it disengaging at ~25 mph since at least then.

It looks like the Honda ACC system disables at ~21.5 mph and openpilot doesn't want to be enabled while the car's cruise control is disabled so it disengages.

[image]

jedberg, 2021-06-24:
@sshane I've only had my Comma2 for six weeks, so definitely something changed between then and now.  Look at the one from 0c769fcdc1652c86|2021-06-09--11-14-49. It clearly keeps cruise engaged all the way to 0mph.  It's even moving the wheel until then.

Also, the first week I had OP I was impressed that it could stop all the way to 0 and even showed it to my brother in law.  So maybe the change happened in .0.8.4, but something definitely changed since I got it (at .0.8.3).

The current behavior is far more dangerous and not chill than when I got it six weeks ago. :)

sshane, 2021-06-24:
Is there another recent route to look at with it stopping to 0? That one's older than 2 weeks so the logs were deleted. Next time, make sure to press "preserve" in useradmin so you can keep the logs around, or I can do that for the next route.

tiki-bug, 2021-06-24:
I saw this same behavior, happened immediately after upgrading from 0.8.4 to 0.8.5 on 6/13/21.  Car used to brake and steer all the way to 0, so long as it didn't send any acceleration commands (if it did, that would cancel OP if below 22mph).  Cancel all OP at 22mph regardless of other states is pretty much stock HondaSense behavior and definitely more dangerous.

Device: Comma 2
Dongle ID: 0a96f86fcfe35964
Version: 0.8.5
Car make/model: 2020 Honda Odyssey

I don't have a route from the 0.8.4 preferred behavior unfortunately, or at least not one that I can see since I don't have prime, but there should be one since I did about 15 hours of driving between 6/11/21 and 6/12/21 which would be inside the 2 week window for unpreserved routes. I could probably get the old code loaded on my device if it was needed for testing.

I do have one route preserved for my device from 2/18/20 that probably exhibits the behavior, this would have been around version 0.7.1 or 0.7.2 but it's behaved the same since I got the device in 1/2020 up until installing 0.8.5.  This segment in particular might have it, I would have to open it up later to see if I did that braking or OP did.

Route: 0a96f86fcfe35964\|2020-02-18--14-43-03--135

jedberg, 2021-06-26:
@sshane There is no route less than two weeks old anymore because it broke two weeks ago.  I don't see a "preserve" button in useradmin, or anything like that, even on routes from yesterday.  Is there some documentation I can look at for preserving routes?  Also is there documentation on how to downgrade versions?  I'm going to go back to 0.8.4 and see if that fixes it.  Thanks.

jedberg, 2021-06-27:
I downgraded back to 0.8.3 and that fixed the problem.  The car will now brake down to 0mph again!  Here is a route with a lot of examples:

0c769fcdc1652c86|2021-06-26--15-00-41
Times:
4 min 10 seconds (15:04:40)
5 min 5 seconds (15:05:35)
6 min 45 seconds (15:07:14) <-- Best example
25 min 15 seconds (15:25:46) <-- Another good example
27 min 5 seconds (15:27:35)
29 min 40 seconds (15:30:10) <-- Car pulled out, lead car was stopped, did the right thing

@sshane I'm still not sure how to preserve logs, but hopefully you can do it/show me before these roll off.  These are from today (Saturday) so we should have two weeks before they disappear.

@pd0wm See above for lots of examples of it working as expected after downgrading to 0.8.3.

Thanks!

sshane, 2021-06-27:
I see some of the logs have uploaded, but not all. Make sure you let it sit on WiFi for a bit longer. And the preserve button is in the main list of routes, right next to the explorer button. There's a bug where you need to preserve *after* it uploads all routes, so make sure you don't preserve before that

tiki-bug, 2021-06-28:
@jedberg Here's a screengrab showing the preserve option
[image]

Do we need any more data showing the previous desired behavior or is there enough already posted?

jedberg, 2021-06-28:
Thank you @sshane I was able to find the button. The logs are now uploaded and preserved.

adeebshihadeh, 2021-07-12:
The old behavior should be restored by #21510.

closed 2021-07-12 by adeebshihadeh

nahcyrag, 2021-08-24:
this is not fixed. I just update to 0.8.8 stock. I notice my honda odyssey 2019 would disengage cruise at 15mph. sending my car towards to the lead car. This was not the behavior at 0.8.3

adeebshihadeh, 2021-08-24:
@nahcyrag please open a new bug report and fill out the template