commaai/openpilot #21363: Driver Monitoring updates in 0.8.5 may cause unsafe practices or positioning for normal operation

JRHonda121, 2021-06-21:
**After updating my Comma 2 to OP v0.8.5 and attempting several drives I have noticed a drastic increase in faults and warnings for Driver Monitoring. While driving the same vehicle, in the same programmed seating position, with the same eyewear I have worn for years, OP proceeded to give me DM faults every 5-10 seconds when seated in the normal position. In an attempt to provide some on the spot "Gemba Testing", I tried removing my glasses, this would stop the faults from occurring. I then tried with my glasses on, as usual, leaning forward extremely close to the steering wheel and the Comma 2, which seemed to mitigate the faults as well. Again, I am not sure if this was an intended change or not, but it has made the Comma 2 unusable for me as the driving position and/or not wearing my glasses would be both unsafe and uncomfortable, thus not making my driving chill.**

**Device/Version information (please complete the following information):**
 - Device: Comma 2
 - Dongle ID: d952837313c3e269
 - Route: No routes showing in web interface for some reason
 - Timestamp: The duration of the last drive on June 19th, 2021 where OP was engaged
 - Version: 0.8.5
 - Car make/model: Honda Pilot 2020

pd0wm, 2021-06-22:
Make sure the "Record and Upload Driver Camera" and "Upload Raw Logs" toggles are both turned on. If you have a route with uploaded footage we can check out your issue.

JRHonda121, 2021-06-22:
I will have a drive tomorrow and will try and test.

JRHonda121, 2021-06-23:
Several drives already completed today for review. Not nearly as extreme so these issues may have been exacerbated by gloomy dark weather. Will have at least 2 more trips today as well.

adeebshihadeh, 2021-06-23:
Post the routes here and preserve them in useradmin.

ursubpar, 2021-06-26:
I have this issue as well. It is impossible to drive in the evening with prescription glasses without DM nagging me to look forward.

adeebshihadeh, 2021-06-26:
@ursubpar please enable the upload raw logs and record driver camera toggles, so we can look into this.

v3lasco, 2021-06-27:
I am also having this same issue and it seems to happen in the evening. I have to either look directly at the device or remove my glasses for it to not nag at me to pay attention since 0.8.5 was installed. The raw logs and driver camera toggle are both on for my Comma 2.

pd0wm, 2021-06-27:
@v3lasco What is your dongle id? Any specific route we should look into?

v3lasco, 2021-06-27:
b7dd7fa08e742ca4 - The last few Sundays evenings, EST, should have a drive where it happened and tonight somewhere between 8:30 and 9:30 PM EST I will be doing a drive.

nelsonjchen, 2021-06-29:
Just wrote up this quick guide to set up the data capture needed for comma.ai . Reporters do not require Comma Prime. However, you do need to post a recording [before the 3 day non-prime upload pruner trims the recordings away before comma.ai staff can preserve them](https://comma.ai/shop/products/comma-prime-sim-card#:~:text=Your%20drives%20are%20stored%20on%20Explorer%20for%C2%A01%C2%A0year%C2%A0(only%203%20days%20without%20prime)).

So, before your joyrides (or non-joyrides) at night, whenever or wherever this issue comes up:

1. Open Settings:
[image]
2. Record your Dongle ID: 
[image]
3. Make sure these two toggles are enabled: 
[image]

**After the incident(s), post your dongle ID here and a rough time frame. You may need to give your device a bit of time to upload.**

If you want to be more precise about the specific drive and/or validate you have the incident recorded for reporting, you can do this: 

1. Go to https://my.comma.ai/ and login with the same *type* of account you paired your device to: 

2. Locate the affected drive, select "More Info", and "View in useradmin".
[image]

3. Copy and paste the drive identifier: 
[image]

cantonese-yan, 2021-06-29:
I tool notice this change of behavior. It seems the false-positive occurs way more often at night when the IR sensor is active.

cydia2020, 2021-06-29:
> I too notice this change of behavior. It seems the false-positive occurs way more often at night (maybe related to the IR sensor?)

#21349

HaraldSchafer, 2021-06-30:
https://github.com/commaai/openpilot/pull/21442

closed 2021-06-30 by HaraldSchafer

BadOlives, 2021-07-01:
Having the same issues as well.

y-a-n-w-a-n-g, 2021-07-01:
This might not matter now since dm has been reverted, but leaving an anecdote that might help figure out the false positives. I was getting this same issue with dm at night. After trying a few different head positions, what got dm to stop bugging me was to keep my eyes and face fixed towards the spot between my comma 2 and the road in front of me. So on a straight road, this would be as if I were looking where a car slightly ahead in the right lane would be.