commaai/openpilot #21368: openpilot unavaiable waiting for controls to start

YJCITA, 2021-06-22:
my car is SKODA SUPERB 2018
when I use v0.8.1， it works ok，but when I use v0.8.4 and v0.8.5, there is some error show on UI:
**openpilot unavaiable 
waiting for controls to start**

I donot know why?

pd0wm, 2021-06-22:
Please fill out the bug report template:
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

adeebshihadeh, 2021-06-22:
The Skoda Superb wasn't supported in 0.8.1. Your car likely never worked on stock openpilot.

adeebshihadeh, 2021-06-25:
Now that we do have support for the Superb, it should be straightforward to get your car supported. Check out https://discord.comma.ai for help getting it supported with the latest code.

closed 2021-06-25 by adeebshihadeh

YJCITA, 2021-07-07:
sorry, I replay so late.
> 
> * Device: EON+balck panda
> * Dongle ID: c6a4ca2ac54533cf
> * Route: c6a4ca2ac54533cf|2021-07-07--06-49-19--0
> * Timestamp: al the time
> * Version: v0.8.5
> * Car make/model: SKODA SUPERB 2018

jyoung8607, 2021-07-07:
@YJCITA, since this is a community-supported car, we can help you at https://discord.comma.ai in the #volkswagen-audi channel (that's the port that supports VW, Audi, Skoda, SEAT, etc).

YJCITA, 2021-07-08:
> @YJCITA, since this is a community-supported car, we can help you at https://discord.comma.ai in the #volkswagen-audi channel (that's the port that supports VW, Audi, Skoda, SEAT, etc).

ths very much
But ,I can not visit this web " https://discord.comma.ai ". 
“#volkswagen-audi channel” is some web data or code，or something else?