commaai/openpilot #21370: 2021 GLI ISSUE

Tdgelo, 2021-06-22:
Hey guys!

Having trouble re-installing openpilot. 

I have a 2021 GLI. It was working initially but not well. It kept saying my calibration was invalid. I would reset and the same issue would pop up. It also would reset itself when I clicked the driver camera preview. 

I uninstalled openpilot and am now trying to reinstall and I just get the spinning wheel and then after a while it goes back to the “getting started” page. Thoughts???

Thanks and I’m sorry if this is the wrong area.

adeebshihadeh, 2021-06-22:
Calibration invalid means you need to remount the device, i.e. uninstall wouldn't fix that. Refer to [these instructions for installing](https://github.com/commaai/openpilot#installation-instructions). If you're still having trouble, join our community discord.

closed 2021-06-22 by adeebshihadeh

Tdgelo, 2021-06-22:
Hey thanks, I did remount and the same issue occurred. 

So I tried uninstalling and now I can’t reinstall.