commaai/openpilot #21373: 2021 GLI

Tdgelo, 2021-06-22:
I uninstalled openpilot on comma two and now it will not reinstall 

**Describe the bug**

**How to reproduce or log data**

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

**Additional context**

adeebshihadeh, 2021-06-22:
I responded to #21370. Our github issues are not for software support, in fact we provide no software support. If there's a bug somewhere, you've provided nothing for us to look into. I just installed openpilot on a fresh device and had no issues.

closed 2021-06-22 by adeebshihadeh

Tdgelo, 2021-06-23:
@adeebshihadeh are you mad at me?