commaai/openpilot #21376: Steeringwheel shaking Issue 2021 Acura

ptajml, 2021-06-23:
**Describe the bug**

**How to reproduce or log data**

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

**Additional context**

adeebshihadeh, 2021-06-23:
Please provide more information and fill out the template

ptajml, 2021-06-23:
> 
> 
> Please provide more information and fill out the template

Sorry i pressed enter while filling out the template

pd0wm, 2021-06-23:
Your PID controller is tuned too aggressively. I suggest lowering kP.

[image]

ptajml, 2021-06-23:
Is there any instructions on lowering kP value?

adeebshihadeh, 2021-06-23:
Here's where the parameters are defined for your car - https://github.com/commaai/openpilot/blob/master/selfdrive/car/honda/interface.py#L338. Join the #tuning channel on our [discord](https://discord.comma.ai) for some help. If you come up with some better values, submit them as a PR.

closed 2021-06-23 by adeebshihadeh