commaai/openpilot #21425: OP v0.8.6 run webcam error： no panda

YJCITA, 2021-06-28:
**Describe the bug**

run webcam , UI show no panda

**How to reproduce or log data**

OP version V0.8.6
[op_maneger_output.txt](https://github.com/commaai/openpilot/files/6724570/op_maneger_output.txt)
[image]

**Expected behavior**

OP should run on PC

**Additional context**

Operating system: ubuntu 20.04

adeebshihadeh, 2021-06-28:
Our issues are not for help getting started. Check out the community discord at https://discord.comma.ai for help getting setup. However, for this issue, ensure your udev rules are setup properly. See https://github.com/commaai/panda#python.

closed 2021-06-28 by adeebshihadeh

visionandy, 2022-10-10:
same problem here, have you solved it?