commaai/openpilot #21428: ui interface display error "openpilot Unavailable,Waiting for controls to start  "

LIHaoran-good, 2021-06-28:
**Describe the bug**

I use the webcam tool to run openpilot on PC and Vehicle . We refer to the [guideline](https://github.com/commaai/openpilot/tree/master/tools/webcam) and connected the PC and vehicle(Using Nissan DBC to make the virtual signal) to two web cameras and Panda .When input "PASSIVE=0 NOSENSOR=1 USE_WEBCAM=1 ./manager.py", the webcam view can be displayed , but displays error "openpilot Unavailable,Waiting for controls to start  ", How to solve it ? Thanks.

**How to reproduce or log data**
When input "PASSIVE=0 NOSENSOR=1 USE_WEBCAM=1 ./manager.py"

**Expected behavior**
Openpilot can run normal on vehicle

**Additional context**

Operating system: [e.g. Ubuntu 20.04]

adeebshihadeh, 2021-06-28:
As we've said in your previous issues, our issues are not for help getting started. Check out the community discord at https://discord.comma.ai for help getting setup.

closed 2021-06-28 by adeebshihadeh

zeroyang0809, 2023-02-19:
hello do you fix your problem? 
i meet the same question
could you share your steps to fix it?
THX