commaai/openpilot #21448: Toyota Corolla Hybrid not showing engine fw

VirtuallyChris, 2021-07-01:
User's car matches all existing fw in values.py, but their logs don't display the engine fw so it won't fingerprint. Is there some fingerprinting bug, or is this an isolated issue with their car?
[image]
839ded97d6d693b0|2021-03-07--17-41-13

adeebshihadeh, 2021-07-02:
This is unlikely a software issue. Many Toyotas only use FW fingerprinting on latest release with no issue.

jyoung8607, 2021-07-04:
When it's consistently the ECU only, we've seen this before when another aftermarket device beats openpilot to the punch opening a ECU diagnostic session at startup. Here are some examples:

* Insurance analytics dongle
* Corporate fleet data recorder/location tracking device
* Battery electric vehicle data logger (for route/charge planning)
* [PolarFIS](https://www.autopolar.net/product/polar-fis-mqb/) or similar add-on display
* Piggyback engine tuning device ([JB4](https://burgertuning.com/collections/jb4) or similar)
* Track data logger

closed 2021-07-07 by adeebshihadeh