commaai/openpilot #21454: Disengagement - Vehicle parameters - Comma-2 v0.8.5 - Prius Prime

charlesmorris5, 2021-07-01:
Experienced numerous disengagements, red screen and alarm regarding vehicle parameters, after ~3 hours of constant driving.  Once started, repeated at random intervals of ~1-10 minutes.  Stopped after unplugging the Comma-2, and plugging it back in (turned it off and on again).  Errors started again after another ~3 hours.  This repeated a few days later on the return trip as well.

A few other users noted similar issues in the Facebook group (https://www.facebook.com/groups/706398630066928).  Different car models seemed to be involved, but Comma-2 with 0.8.5 was consistent.

brianhaugen2, 2021-07-02:
Can confirm similar issue also with 0.8.5
Had a UI reset (but openpilot stayed engaged) and UI was loading for about 10 minutes.
After UI loaded received red screen and alarm with disengagement.
Similar scenario of driving for multiple hours on a long drive.

adeebshihadeh, 2021-07-02:
Please fill out the bug report template so we can look into this.

charlesmorris5, 2021-07-02:
> Please fill out the bug report template so we can look into this.

Sorry, newb here.  I thought this was a bug report.  Please advise link or process.

adeebshihadeh, 2021-07-02:
Here's the template. It's the text that's automatically filled in when you open an issue. 

**Describe the bug**

**How to reproduce or log data**

**Expected behavior**

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

**Additional context**

charlesmorris5, 2021-07-02:
Huh, I didn't see any of that.

Describe the bug:
Experienced numerous disengagements, red screen and alarm regarding vehicle parameters, after ~3 hours of constant driving. Once started, repeated at random intervals of ~1-10 minutes. Stopped after unplugging the Comma-2, and plugging it back in (turned it off and on again). Errors started again after another ~3 hours. This repeated a few days later on the return trip as well.

A few other users noted similar issues in the Facebook group (https://www.facebook.com/groups/706398630066928). Different car models seemed to be involved, but Comma-2 with 0.8.5 was consistent.

How to reproduce or log data:
~3 hours of constant driving.

Expected behavior:
Not disengage

Device/Version information (please complete the following information):
Device: Comma Two
Dongle ID:  4879b9286bb8bd90

Route(s):
https://my.comma.ai/useradmin/?onebox=4879b9286bb8bd90%7C2021-06-29--07-23-31
https://my.comma.ai/useradmin/?onebox=4879b9286bb8bd90%7C2021-06-29--12-23-26
https://my.comma.ai/useradmin/?onebox=4879b9286bb8bd90%7C2021-06-30--07-22-00

Timestamp: 
29 Jun 10:00-11:00, 15:25-1600 (numerous)
30 Jun 10:00, 11:09, 11:10, 11:30 (started adjusting behavior to reduce incidents; reboot Comma or rest stop)

Version: v0.8.5

Car make/model: Toyota Prius Prime Advanced 2017

Additional context:
-No errors shown by car displays
-Stopped errors after rebooting Comma-2
-Time fluctuated, averaging about 3 hours of constant driving before errors started, then frequent

pd0wm, 2021-07-02:
Can you make sure the toggle "Upload raw logs" is on and your device is connected to wifi? The full logs for those drives are not uploaded.

charlesmorris5, 2021-07-02:
I just plugged it in, connected to wi-fi and toggled upload raw logs to on. I'm sure the upload will take some time...

pd0wm, 2021-07-07:
Don't see any uploaded rlogs, they might have been deleted already due to a full disk.

I did some debugging using the qlogs. I'm seeing both stiffness and steer ratio go down while driving at the same speed for a few hours. At this constant speed both values changing at the same results in the same curvature rate, so it doesn't influence driving but will eventually trigger the sanity check.

Usually due to variations in speed the variables converge to the right value, however the params learner problem is under constrained at single speed, so you run the risk of the values moving away together. I'm not sure if this is made worse by the sloppy angle sensor in the prius or if it's something else that's causing you to have these issues.

I'll see if I can reproduce the problem in simulation, and will consider adding some regularization where the values will converge back to the initial values if there is no good data to learn from.

nuwandavek, 2022-02-14:
Should be fixed with https://github.com/commaai/openpilot/pull/23726

closed 2022-02-14 by nuwandavek