commaai/openpilot #21464: Logs should contain blank frames if the frame is dropped.

HaraldSchafer, 2021-07-02:
If a frame is not received from the camera for any reason, a blank frame should be inserted into the video stream and a flag should be raised. This ensures synchronization of the camera stream with other logs and other camera streams.

adeebshihadeh, 2021-10-11:
Did we decide that this is unnecessary if the frame drops are rare enough?

HaraldSchafer, 2021-10-11:
Yes, that is ok if it's rare enough (less than 1% of segments), but:

- Frame drops should still be flagged
- Frame drops cannot cause synchronization issues between the camera streams for the rest of the route, everything should be resolved and re synchronized by the start of the next segment.

gregjhogan, 2021-10-11:
Why do we wait for the frame after the skipped frame to detect skipped frames?  Is it a bad idea to clock out blank frames (flagged as such) if we don't get a new frame within specific timing constraints, and then ignore real frames that come in very late?

adeebshihadeh, 2021-11-29:
@HaraldSchafer can we close this now?

gregjhogan, 2021-11-29:
I thought this was a good idea until I realized that you have to send the blank frames through the encoder which also sometimes drops frames, at which point this doesn't seem practical

closed 2021-11-30 by HaraldSchafer