commaai/openpilot #21470: comm issue selfdrive/boardd/panda.cc: Transmit buffer full

mogorman, 2021-07-02:
**Describe the bug**

I was driving on highway, nothing out of ordinary then got a red screen of death.  I believe it complained about communication processes but I didn't get a chance to read it as I immediately canceled OP and took over driving. Unclear how to reproduce.

```
selfdrive/boardd/panda.cc: Transmit buffer full                                                                                                                                                                                                
selfdrive/boardd/panda.cc: Transmit buffer full                                                                                                                                                                                                
selfdrive/boardd/panda.cc: Transmit buffer full                                                                                                                                                                                                
selfdrive/boardd/panda.cc: Transmit buffer full                                                                                                                                                                                                
commIssue - valid: {'jvePilotUIState': True, 'deviceState': True, 'pandaState': True, 'modelV2': True, 'liveCalibration': True, 'driverMonitoringState': True, 'longitudinalPlan': True, 'lateralPlan': False, 'liveLocationKalman': True, 'man
agerState': True, 'liveParameters': True, 'radarState': True, 'roadCameraState': True, 'driverCameraState': True} - alive: {'jvePilotUIState': True, 'deviceState': True, 'pandaState': True, 'modelV2': True, 'liveCalibration': True, 'driver
MonitoringState': True, 'longitudinalPlan': True, 'lateralPlan': True, 'liveLocationKalman': True, 'managerState': True, 'liveParameters': True, 'radarState': True, 'roadCameraState': True, 'driverCameraState': True}                       
selfdrive/boardd/panda.cc: Transmit buffer full                                                                                                                                                                                                
selfdrive/boardd/panda.cc: Transmit buffer full                                                                                                                                                                                                
selfdrive/boardd/panda.cc: Transmit buffer full                                                                                                                                                                                                
selfdrive/boardd/panda.cc: Transmit buffer full                                                                                                                                                                                                
selfdrive/boardd/panda.cc: Transmit buffer full                                                                                                                                                                                                
selfdrive/boardd/panda.cc: Transmit buffer full                                                                                                                                                                                                
selfdrive/boardd/panda.cc: Transmit buffer full                                                                                                                                                                                                
selfdrive/boardd/panda.cc: Transmit buffer full                                                                        
selfdrive/boardd/panda.cc: Transmit buffer full                                                                                                                                                                                                
selfdrive/boardd/panda.cc: Transmit buffer full                                                                                                                                                                                                
selfdrive/boardd/panda.cc: Transmit buffer full                                                                                                                                                                                                
selfdrive/boardd/panda.cc: Transmit buffer full                                                                                                                                                                                                
selfdrive/boardd/panda.cc: Transmit buffer full                                                                                                                                                                                                
selfdrive/boardd/pigeon.cc: Storing almanac in ublox flash                                                                                                                                                                                     
selfdrive/boardd/panda.cc: Transmit buffer full                                                                                                                                                                                                
selfdrive/boardd/panda.cc: Transmit buffer full                                                                                                                                                                                                
killing camerad                                                                                                                                                                                                                                
sending signal 2 to camerad                                                                                                                                                                                                                    
selfdrive/boardd/panda.cc: Transmit buffer full                                                                                                                                                                                                
killing clocksd                                                                                                                                                                                                                                
sending signal 2 to clocksd                                                                                                                                                                                                                    
killing dmonitoringmodeld                                                                                                                                                                                                                      
sending signal 2 to dmonitoringmodeld                                                                                                                                                                                                          
killing logcatd                                                                                                                                                                                                                                
sending signal 2 to logcatd                                                                                                                                                                                                                    
killing loggerd                                                                                                                                                                                                                                
sending signal 2 to loggerd                                                                                                                                                                                                                    
[... cut ...]
 - Version: 2034ff62e8815f779500e34ab927c4e0e8400ffd
 - Car make/model: chrysler pacifica 2021

**Additional context**
This happened on my own fork. https://github.com/mogorman/openpilot/tree/mog_beta I realize you probably will just close because of that but wanted to report as it seems odd that sending an mqtt message every thirty seconds could make the c2 so unstable.  I also have been hearing a lot of complaints about similiar behavior in discord but not seeeing any bug reports. so thought id be the first.

mogorman, 2021-07-02:
2 errors around time of error in my swaglog
```
swaglog.0000006266:{"ctx": {"dongle_id": "021c3223d895b161", "version": "0.8.5(v3.4.1)", "dirty": false, "device": "eon"}, "exc_info": "Traceback (most recent call last):\n  File \"/data/openpilot/selfdrive/thermald/thermald.py\", line 236, in thermald_thread\n    network_strength = HARDWARE.get_network_strength(network_type)\n  File \"/data/openpilot/selfdrive/hardware/eon/hardware.py\", line 269, in get_network_strength\n    out = subprocess.check_output('dumpsys connectivity', shell=True).decode('utf-8')\n  File \"/data/data/com.termux/files/usr/lib/python3.8/subprocess.py\", line 411, in check_output\n    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,\n  File \"/data/data/com.termux/files/usr/lib/python3.8/subprocess.py\", line 512, in run\n    raise CalledProcessError(retcode, process.args,\nsubprocess.CalledProcessError: Command 'dumpsys connectivity' died with <Signals.SIGUSR2: 12>.", "level": "ERROR", "levelnum": 40, "name": "swaglog", "filename": "thermald.py", "lineno": 239, "pathname": "/data/openpilot/selfdrive/thermald/thermald.py", "module": "thermald", "funcName": "thermald_thread", "host": "localhost", "process": 2277, "thread": 547805978648, "threadName": "MainThread", "created": 1625223994.8388598, "msg$s": "Error getting network status", "id": "8516bc7e72f44e1a9d0b697c882fb8a4"} 
swaglog.0000006402:{"ctx": {"dongle_id": "021c3223d895b161", "version": "0.8.5(v3.4.1)", "dirty": false, "device": "eon"}, "level": "ERROR", "levelnum": 40, "name": "swaglog", "filename": "controlsd.py", "lineno": 656, "pathname": "/data/openpilot/selfdrive/controls/controlsd.py", "module": "controlsd", "funcName": "step", "host": "localhost", "process": 2449, "thread": 548544614424, "threadName": "MainThread", "created": 1625235699.895005, "msg$s": "commIssue - valid: {'jvePilotUIState': True, 'deviceState': True, 'pandaState': True, 'modelV2': True, 'liveCalibration': True, 'driverMonitoringState': True, 'longitudinalPlan': True, 'lateralPlan': True, 'liveLocationKalman': True, 'managerState': True, 'liveParameters': True, 'radarState': True, 'roadCameraState': True, 'driverCameraState': True} - alive: {'jvePilotUIState': True, 'deviceState': True, 'pandaState': True, 'modelV2': True, 'liveCalibration': True, 'driverMonitoringState': True, 'longitudinalPlan': True, 'lateralPlan': True, 'liveLocationKalman': True, 'managerState': True, 'liveParameters': False, 'radarState': True, 'roadCameraState': True, 'driverCameraState': True}", "id": "92a5d67d22ce49598fdcb187846387f6"}
```

pd0wm, 2021-07-02:
Saw a lot of cellular activity around the event, which is causing lags and a disengagement. We're working on a solution. See https://github.com/commaai/openpilot/issues/20872 https://github.com/commaai/openpilot/issues/19874

TheWizard1328, 2021-07-02:
I am getting similar issue.
As of Tue June 29 my comma stopped working.
I start van, 2020 Pacifica S Launch Edition, and when I engage the ACC the steering wheel in top right corner does not appear. When I try to set Max speed I get this error on my comma: "Openpilot Unavailable. Communication issues between processes."
and on my cluster: "Adaptive Cruise Control (ACC) Cancelled"

pd0wm, 2021-07-02:
That sounds like a different issue, please create a new bug report and fill out the template.

closed 2021-07-04 by pd0wm

tecandrew, 2021-07-04:
Just realized been seeing this on Elantra 2021, but never bothered looking at `tmux a`. A reboot seemed to have solved this initially, but now it's occurring more often and a reboot does nothing.

It only occurs during startup.

mogorman, 2021-07-07:
@pd0wm I don't understand why this and the related issues were closed. still having issues. is the recommended fix to send back my c2 to comma? It seems related to 0.8.5 upgrade to me

ubiaman, 2021-07-26:
hello i am getting the communication issue between processes it started after 8.6 update and i have tried rebooting  iw will not allow me to use autopilot

TheWizard1328, 2021-07-27:
Mine was doing it on 0.8.5. But stopped doing it after a bought a new USB C cable from The Source.