commaai/openpilot #21479: 2019 Kia Niro PHEV - No steering under 32 MPH

pauste, 2021-07-04:
**Describe the bug**: When system is active (ACC turned on) if the driving speed (not the set speed) goes below 32 MPH, the steering control will disengage with out warning. When driving under 32 MPH then moving the set speed from 29 to 32 MHP, the system can violently (quickly turning right or left) as the system assumes control as the car transitions from 31 to 32 MPH. This can be quite alarming and renders the system unsafe under those operations. Rolling the set speed from above 32 MPH to below 32 results in a benign disengagement of the steering (i.e. just stops controlling the steering but doesn’t do so in a erratic behavior). This behavior makes the system not usable under 32 MPH and dangerous in freeway stop-n-go traffic. 

I’ve tried both Circuit-pro fork “hkg_085” and Donfyffe, also with v0.8.5, and both have the same 32 MPH issue described above.

**How to reproduce or log data**: As stated above. 

**Expected behavior**: I would like the system to control the steering wheel once ACC is set in the car and allow the car’s steering wheel to be controlled from 0 MPH, there by allowing the software to function in stop-n-go traffic. At the very least there should be a warning to NOT use the system in stop-n-go traffic especially if the car speeds bounces above and below 32 MPH. It would also be nice to have something on the C2 display showing when C2 is actively controlling the steering wheel. 

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: cd5179e1
 - Route: ?? Not using comma services so my.comma.ai/useradmin is blank
 - Panda FW: 46fda13fee69be84
 - Timestamp: Anytime the conditions are ACC engaged and speed is set below 32 MPH
 - Version: v0.8.5 from “Circuit-Pro” branch “hkg_085”
 - Car make/model: Kia Niro Plug-in Hybrid 2019

**Additional context**

jyoung8607, 2021-07-04:
There are many instances where older cars have minimum steering speeds, enforced by the vehicle's EPS firmware, that are reduced or eliminated in newer versions of that same car. It sounds like your car is one of those.

openpilot has support to cleanly handle transitions across the speed threshold, but it has to be enabled specifically for your car. See for example the [Hyundai Ioniq 2019 vs 2020](https://github.com/commaai/openpilot/blob/32d5239912b592078faf50628e0f08b07a0eb755/selfdrive/car/hyundai/interface.py#L127-L128), which is also documented in the README.

The 2019 Kia Niro isn't yet a supported car in upstream openpilot. It appears it's been added in some forks that you've been using, but the fork maintainer neglected to add the minimum steer speed. You have two choices:

* Work with comma to have upstream support added for your car (preferred, fixes it for everyone)
* Work with your fork maintainer to fix the bug in just their fork

For future reference: while this one happens to be easy to figure out, if you're not running comma software and not even uploading logs to comma's backend, you should work with your fork maintainer rather than open issues here.

jyoung8607, 2021-07-04:
As a second and only partially related issue, I'm wondering why openpilot doesn't show a LKAS Fault in this situation. On most cars, if we're trying to steer and the EPS is rejecting our request, we detect that and alert the driver. From reading your issue, it doesn't sound like openpilot is detecting that. If the HKG EPS makes this condition detectable, and openpilot isn't detecting it, ISTM this is a separate bug that should also be fixed. But, there's no way to tell without logs from an affected route.

pauste, 2021-07-04:
Let me know what I need to do to Work with comma to have upstream support added for my car. 

BTW: Before moving to the fork I’m using I originally tried using standard OP v0.8.5 but got a message ”CAN ERROR: check connections”..

jyoung8607, 2021-07-04:
@pauste, DM me on Discord and I'll see if I can help.

adeebshihadeh, 2021-07-05:
As @jyoung8607 noted, your car isn't supported, but we'd be interested in getting your car supported once a PR is opened.

closed 2021-07-05 by adeebshihadeh

Araziel222, 2026-10-06:
hello i wanted to know this issue was able to be resolved somehow? it seems like it was closed and i wanted to know if there was ever any sort of update on this.

pauste, 2026-11-01:
No update on this. I think John wasn't able resolve it on my Niro. I no longer have the Niro to test and I think John (who I was working with) doesn't do these updates anymore.

PR #21481 (2021-07-08): HKG: 2019 Kia Niro PHEV
**Checklist**
- [x] added to README
- [x] test route added to [test_routes.py](https://github.com/commaai/openpilot/blob/master/selfdrive/test/test_routes.py)
- [x] route with openpilot: 173219cf50acdd7b|2021-07-05--10-27-41

Although this was never a bug in upstream openpilot, this PR effectively resolves #21479.

**Important note:** This removes conflicting UDS fingerprint entries for OPTIMA_H, because they appear to be erroneous:
* A single OPTIMA_H FPv2 entry was added in #1682, and never touched again
* They are an exact 100% collision with this Kia Niro PHEV
* If  [... cut]