commaai/openpilot #21480: Model training code

nadashin, 2021-07-04:
I have a question.

Which part of this repository has the code for training the model?
Is it in a different repository?

I am opening this issue after not getting email answer from Comma and openpilot's top committers.

adeebshihadeh, 2021-07-04:
This repo doesn’t have any training code. The training code is not public.

closed 2021-07-04 by adeebshihadeh

beeman, 2021-07-05:
'Open'pilot...

[image]

nadashin, 2021-07-07:
*Email Reply*

Chris McCammon (comma.ai)

Jul 6, 2021, 19:46 MST
Hi Nadashin,

We won't be open sourcing the training infrastructure. Doing so would only help large corporations and not be helpful to any individuals trying to push ML further.

You need a 100+ thousand datacenter to train at any reasonable speed.

Best,
Chris