commaai/openpilot #21500: Steering Wheel is "Shaking"

ffhspa, 2021-07-06:
**Describe the bug**
Steering wheel is shaking during a Lane change and in some other situations. 

**How to reproduce or log data**

**Expected behavior**
Steering wheel should not shake

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: 71753668efc54f63
 - Route: 71753668efc54f63|2021-07-06--13-32-51,   | 71753668efc54f63\|2021-07-06--11-46-51, 71753668efc54f63|2021-07-06--11-27-54, 71753668efc54f63|2021-07-06--11-18-31
-- | --
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: 0.8.5 release2
 - Car make/model: Honda Civic Diesel 

**Additional context**

adeebshihadeh, 2021-07-06:
This is a result of poor tuning. Check out the #tuning channel at https://discord.comma.ai for help tuning your car.

closed 2021-07-06 by adeebshihadeh

ffhspa, 2021-07-06:
@adeebshihadeh i havent tuned anything. Its stock release2