commaai/openpilot #21508: communication error on driving

TheWizard1328, 2021-07-07:
**Describe the bug**

The majority of  times I start my car the wheel in corner of the openpilot does not come up and displays the message "openpilot unavailable. waiting for controls to start"

**How to reproduce or log data**
Start car and try to use openpilot not as a dashcam

**Expected behavior**
Open pilot should make driving chill.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: 8a84e5027e568485
 - Route: 8a84e5027e568485|2021-07-07--18-08-44 & 8a84e5027e568485|2021-07-07--19-05-48
 - Timestamp: when I hit the ACC+ to engage ACC
 - Version: 0.8.5
 - Car make/model: 2020 Chrysler Pacifica S Launch Edition

**Additional context**

adeebshihadeh, 2021-07-07:
The full logs aren't uploaded (connect your device to wifi and enable the upload raw toggle), but controls starts within a second in both the routes you posed. UI waits 10 seconds before showing the "Waiting for controls to start" alert, so I'm not sure why it would show the alert in that case. Are you sure it happened on those routes?

TheWizard1328, 2021-07-07:
All toggles are on except for Lanaless(Alpha) and right hand drive.
What keeps happening is 
1) I activate ACC
2) Set max speed with ACC+
Then I get the message "openpilot unavailable. waiting for controls to start" on the comma and "Adaptive Cruse Control (ACC) Cancelled" on my cluster screen. 
[image]

Two new drives:
8a84e5027e568485|2021-07-07--22-46-45
8a84e5027e568485|2021-07-07--22-44-28

Screen shot images attached.
[image]
[image]
[image]
[image]

TheWizard1328, 2021-07-08:
so i took  drive this morning. ACC engaged fine, showed the orange message for no steer below kmh. but then it turned off and my cluster showed the ACC unavailable, service required.
Drive ID: 8a84e5027e568485|2021-07-08--15-17-01
Then I switched to Dashcam mode so I could use my stock ACC. Part way through the drive I saw an LKAS warning on my comma, but ACC was still working fine and still engaged, then the steering wheel started showing up when ACC was in control and disappearing when I hit the breaks.
Drive ID: 8a84e5027e568485|2021-07-08--15-41-16

TheWizard1328, 2021-07-09:
Just took 2 drives:
8a84e5027e568485|2021-07-09--18-00-44
and 
8a84e5027e568485|2021-07-09--17-46-25
and all went great had full comma steer support for both drives

TheWizard1328, 2021-07-11:
been driving with stock OP v0.8.5 seems to work good about 95% of the time. One my last drive last night I got a loud beeping warning du to lost communication at about 22:02:50 during drive 8a84e5027e568485|2021-07-11--03-48-48

TheWizard1328, 2021-07-12:
I do not know if this is related or not but could you check time stamp 16:56 during this drive.
8a84e5027e568485|2021-07-11--22-55-43
Had an odd error that cause my comma to flash a warning and disengage till I was able to restart the car.
Note I am using the prerelease of 0.8.6 as so far it is the only branch that seems to work for me right now. Other then the odd LKAS when changing lanes if my blinker turns off too soon.

adeebshihadeh, 2021-11-21:
Is this still an issue?

TheWizard1328, 2021-11-22:
Not that I can think of. IT has been a few months now a several updates.

closed 2021-11-22 by TheWizard1328