commaai/openpilot #21519: Ddisengagement 2 times without warning.

johnlu6518, 2021-07-08:
**Describe the bug**

Did not step on the gas, but disengagement 2 times without warning.

**How to reproduce or log data**
Device: Comma 2
Dongle ID: ec30903bb15db3ad
Route: ec30903bb15db3ad|2021-07-05--22-31-41
Timestamp: Tue, 6 Jul @ 11:05:44 & Tue, 6 Jul @ 11:11:14
Veraion: OP 0.8.5
Car make/model: Lexus IS 2017 200T

[image]

adeebshihadeh, 2021-07-08:
The full logs for that route aren't uploaded. Can you enable the "Upload Raw" toggle and connect your device to wifi?

johnlu6518, 2021-07-08:
Okay, I'm not at home, so I will upload it later.

johnlu6518, 2021-07-08:
Device online and the data uploaded now! Thanks a lot.

sshane, 2022-02-02:
Looks like the car's PCM disabled on its own, which could be caused by an accidental cancel button press, traction control activating or more. Do you have any more recent cases of this?

adeebshihadeh, 2022-02-02:
@johnlu6518 looks like your last routes are from September. Do you still have your device? If not, I think we can close this.

adeebshihadeh, 2022-02-04:
Closing due to inactivity. Feel free to reopen if you still have the issue.

closed 2022-02-04 by adeebshihadeh