commaai/openpilot #21530: Crash in controlsd

kennedyshead, 2021-07-09:
**Describe the bug**

Crash in controlsd
**How to reproduce or log data**
Engage and wait

```
Traceback (most recent call last):
  File "capnp/lib/capnp.pyx", line 1262, in capnp.lib.capnp._DynamicStructBuilder.__setattr__
  File "capnp/lib/capnp.pyx", line 1255, in capnp.lib.capnp._DynamicStructBuilder._set
  File "capnp/lib/capnp.pyx", line 747, in capnp.lib.capnp._setDynamicField
capnp.lib.capnp.KjException: Tried to set field: 'gas' with a value of: '0.4994253880211296' which is an unsupported type: '<class 'numpy.float64'>'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/data/openpilot/selfdrive/manager/process.py", line 36, in launcher
    mod.main()
  File "/data/openpilot/selfdrive/controls/controlsd.py", line 673, in main
    controls.controlsd_thread()
  File "/data/openpilot/selfdrive/controls/controlsd.py", line 667, in controlsd_thread
    self.step()
  File "/data/openpilot/selfdrive/controls/controlsd.py", line 657, in step
    actuators, lac_log = self.state_control(CS)
  File "/data/openpilot/selfdrive/controls/controlsd.py", line 455, in state_control
    actuators.gas, actuators.brake, self.v_target, self.a_target = self.LoC.update(self.active, CS, self.CP, long_plan)
  File "capnp/lib/capnp.pyx", line 1264, in capnp.lib.capnp._DynamicStructBuilder.__setattr__
  File "capnp/lib/capnp.pyx", line 234, in capnp.lib.capnp.KjException._to_python
AttributeError: 'NoneType' object has no attribute 'type'
Process controlsd:
Traceback (most recent call last):
  File "capnp/lib/capnp.pyx", line 1262, in capnp.lib.capnp._DynamicStructBuilder.__setattr__
  File "capnp/lib/capnp.pyx", line 1255, in capnp.lib.capnp._DynamicStructBuilder._set
  File "capnp/lib/capnp.pyx", line 747, in capnp.lib.capnp._setDynamicField
capnp.lib.capnp.KjException: Tried to set field: 'gas' with a value of: '0.4994253880211296' which is an unsupported type: '<class 'numpy.float64'>'
```

HaraldSchafer, 2021-07-09:
https://github.com/commaai/openpilot/pull/21536

HaraldSchafer, 2021-07-09:
The cause is the lack float casting here
https://github.com/commaai/openpilot/blob/35a5aae334e2409c432499a868fae69bd6ceb285/selfdrive/car/tesla/interface.py#L12

closed 2021-07-09 by HaraldSchafer