commaai/openpilot #21554: Black screen after install on fresh C2

brianfitzgerald, 2021-07-11:
**Describe the bug**

My comma two stopped working a couple weeks ago - it will power on but once it goes past the Comma logo startup the screen turns black (with the backlight still on). I was able to get into fastboot and do a factory reset via eon-neos, but once it finished installing openpilot it returned to a black screen. Same results with installing dashcam. 

Install logs:

`Pre-checkout found
dinfo 1920x1080
egl version 1.4
egl w 1920 h 1080
gl version OpenGL ES 3.2 V@145.0 (GIT@I0ac3940325)
remote: Enumerating objects: 61, done.
remote: Counting objects: 100% (61/61), done.
remote: Compressing objects: 100% (34/34), done.
remote: Total 40 (delta 23), reused 21 (delta 4), pack-reused 0
Unpacking objects: 100% (40/40), done.
From https://github.com/commaai/openpilot
 * branch            release2   -> FETCH_HEAD
 * [new branch]      release2   -> origin/release2
Branch 'release2' set up to track remote branch 'release2' from 'origin'.
Switched to a new branch 'release2'
HEAD is now at aee971b Hotfix: revert DM model update (#21450)
INFO:root:System time valid
Using interface 'wlan0'
OK
3
3
/data/openpilot/launch_chffrplus.sh: line 35: echo: write error: No such device
/data/openpilot/launch_chffrplus.sh: line 63: /proc/irq/184/smp_affinity_list: No such file or directory
/data/openpilot/launch_chffrplus.sh: line 63: /proc/irq/185/smp_affinity_list: No such file or directory
/data/openpilot/launch_chffrplus.sh: line 63: /proc/irq/186/smp_affinity_list: No such file or directory
pid 209's current scheduling policy: SCHED_OTHER
pid 209's current scheduling priority: 0
pid 209's new scheduling policy: SCHED_FIFO
pid 209's new scheduling priority: 52
pid 210's current scheduling policy: SCHED_OTHER
pid 210's current scheduling priority: 0
pid 210's new scheduling policy: SCHED_FIFO
pid 210's new scheduling priority: 52
pid 319's current scheduling policy: SCHED_OTHER
pid 319's current scheduling priority: 0
pid 319's new scheduling policy: SCHED_FIFO
pid 319's new scheduling priority: 52
pid 320's current scheduling policy: SCHED_OTHER
pid 320's current scheduling priority: 0
pid 320's new scheduling policy: SCHED_FIFO
pid 320's new scheduling priority: 52
Result: Parcel(00000000 00000001   '........')
System time valid
getting pilotauth
starting process ui
preimporting selfdrive.locationd.calibrationd
preimporting selfdrive.controls.controlsd
preimporting selfdrive.loggerd.deleter
preimporting selfdrive.monitoring.dmonitoringd
preimporting selfdrive.logmessaged
preimporting selfdrive.pandad
preimporting selfdrive.locationd.paramsd
preimporting selfdrive.controls.plannerd
preimporting selfdrive.controls.radard
preimporting selfdrive.rtshield
preimporting selfdrive.thermald.thermald
preimporting selfdrive.tombstoned
preimporting selfdrive.updated
preimporting selfdrive.loggerd.uploader
manager start
{'environ': environ({'SHELL': '/system/comma/usr/bin/bash', 'TMUX': '/data/data/com.termux/files/usr/tmp//tmux-0/default,1818,0', 'PREFIX': '/usr', 'NUMEXPR_NUM_THREADS': '1', 'GIT_SSH_COMMAND': 'ssh -i /data/gitkey', 'ANDROID_BOOTLOGO': '1', 'EDITOR': 'vim', 'PWD': '/data/openpilot/selfdrive/manager', 'OPENBLAS_NUM_THREADS': '1', 'REQUIRED_NEOS_VERSION': '17', 'SYSTEMSERVERCLASSPATH': '/system/framework/org.cyanogenmod.platform.jar:/system/framework/org.cyanogenmod.hardware.jar:/system/framework/services.jar:/system/framework/ethernet-service.jar:/system/framework/wifi-service.jar', 'EXTERNAL_STORAGE': '/sdcard', 'LD_PRELOAD': 'libsigchain.so', 'HOME': '/data/data/com.termux/files/home', 'AGNOS_VERSION': '0.18', 'LANG': 'en_US.UTF-8', 'PYTHONSTARTUP': '/data/data/com.termux/files/home/.pythonrc', 'LD_SHIM_LIBS': '/system/lib/hw/camera.msm8996.so|libcamera_shim.so', 'VECLIB_MAXIMUM_THREADS': '1', 'ANDROID_CACHE': '/cache', 'QT_QPA_EGLFS_PHYSICAL_WIDTH': '151', 'ANDROID_DATA': '/data', 'ANDROID_STORAGE': '/storage', 'PYTHONPATH': '/data/openpilot', 'TERM': 'xterm-256color', 'TERMINFO': '/system/etc/terminfo', 'VIMINIT': 'source $MYVIMRC', 'TMUX_PANE': '%0', 'ASEC_MOUNTPOINT': '/mnt/asec', 'QT_QPA_EGLFS_HIDECURSOR': '1', 'SHLVL': '1', 'QT_QPA_FONTDIR': '/usr/share/fonts', 'VALGRIND_LIB': '/usr/lib/valgrind', 'ANDROID_ROOT': '/system', 'BOOTCLASSPATH': '/system/framework/core-libart.jar:/system/framework/conscrypt.jar:/system/framework/okhttp.jar:/system/framework/core-junit.jar:/system/framework/bouncycastle.jar:/system/framework/ext.jar:/system/framework/framework.jar:/system/framework/telephony-common.jar:/system/framework/voip-common.jar:/system/framework/ims-common.jar:/system/framework/apache-xml.jar:/system/framework/org.apache.http.legacy.boot.jar', 'PASSIVE': '0', 'LD_LIBRARY_PATH': '/data/phonelibs:/data/data/com.termux/files/usr/lib', 'MYVIMRC': '/data/data/com.termux/files/home/.vimrc', 'QT_QPA_EVDEV_TOUCHSCREEN_PARAMETERS': '/dev/input/event1:rotate=270', 'OMP_NUM_THREADS': '1', 'STAGING_ROOT': '/data/safe_staging', 'PATH': '/usr/local/bin:/data/data/com.termux/files/usr/bin:/data/data/com.termux/files/usr/sbin:/data/data/com.termux/files/usr/bin/applets:/bin:/sbin:/vendor/bin:/system/sbin:/system/bin:/system/xbin', 'ANDROID_PROPERTY_WORKSPACE': '9,0', 'MKL_NUM_THREADS': '1', 'QT_QPA_EGLFS_PHYSICAL_HEIGHT': '74', 'ANDROID_ASSETS': '/system/app', 'OLDPWD': '/data/openpilot', '_': './manager.py', 'DONGLE_ID': '8b94ce9faacf1b02', 'CLEAN': '1'})}
selfdrive/loggerd/bootlog.cc: bootlog to /data/media/0/realdata/boot/2021-07-11--13-40-02.bz2
starting daemon manage_athenad
starting process sensord
[... 10 lines cut ...]
deviceState: Reader was evicted, reconnecting
sensord ui deleter logmessaged pandad thermald tombstoned updated uploader
managerState: Reader was evicted, reconnecting
sensord ui deleter logmessaged pandad thermald tombstoned updated uploader
{"event": "STATUS_PACKET", "count": 0, "pandaState": null, "location": null, "deviceState": {"deviceState": {"freeSpacePercent": 97.56057739257812, "batteryPercent": 97, "batteryStatus": "Discharging", "fanSpeedPercentDesired": 0, "started": false, "usbOnline": false, "startedMonoTime": 0, "thermalStatus": "green", "batteryCurrent": 624080, "batteryVoltage": 4209875, "chargingError": false, "chargingDisabled": false, "memoryUsagePercent": 22, "cpuUsagePercent": 36, "networkType": "wifi", "offroadPowerUsageUwh": 0, "networkStrength": "great", "carBatteryCapacityUwh": 3000000, "cpuTempC": [42.70000076293945, 39.400001525878906, 37.79999923706055, 37.79999923706055], "gpuTempC": [36.0], "memoryTempC": 40.099998474121094, "batteryTempC": -27.5, "ambientTempC": 30.0, "lastAthenaPingTime": 0}, "logMonoTime": 428996382645, "valid": true}}
sensord ui deleter logmessaged pandad thermald tombstoned updated uploader
sensord ui deleter logmessaged pandad thermald tombstoned updated uploader
sensord ui deleter logmessaged pandad thermald tombstoned updated uploader
sensord ui deleter logmessaged pandad thermald tombstoned updated uploader
sensord ui deleter logmessaged pandad thermald tombstoned updated uploader
upload ('boot/2021-07-11--13-40-02.bz2', '/data/media/0/realdata/boot/2021-07-11--13-40-02.bz2') over wifi
{"event": "upload", "key": "boot/2021-07-11--13-40-02.bz2", "fn": "/data/media/0/realdata/boot/2021-07-11--13-40-02.bz2", "sz": 59563}
checking 'boot/2021-07-11--13-40-02.bz2' with size 59563
uploading '/data/media/0/realdata/boot/2021-07-11--13-40-02.bz2'
sensord ui deleter logmessaged pandad thermald tombstoned updated uploader
upload_url v1.3 https://commadata2.blob.core.windows.net/commadata2/8b94ce9faacf1b02/boot/2021-07-11--13-40-02.bz2?se=2021-07-12T18%3A40%3A09Z&sp=cw&sv=2018-03-28&sr=b&sig=7j20GvnEFE%2BAELqdJMLowOlVY8obWn7ogBKGIkcgZms%3D {'x-ms-blob-type': 'BlockBlob'}
sensord ui deleter logmessaged pandad thermald tombstoned updated uploader
{"event": "upload_success", "key": "boot/2021-07-11--13-40-02.bz2", "fn": "/data/media/0/realdata/boot/2021-07-11--13-40-02.bz2", "sz": 59563, "debug": true}
upload done, success=True`

**Device/Version information (please complete the following information):**
 - Device: Comma two
 - Dongle ID: N/A (since install is unavailable)
 - Version: 0.8.5, NEOS version 17
 - Car make/model: Toyota Corolla 2018

adeebshihadeh, 2021-07-11:
Are any of the processes red in that first tmux pane? Can you post the dongle ID (`cat /data/params/d/DongleId`)?

brianfitzgerald, 2021-07-11:
FIle is empty - though I'm connected to my Mac. Will plug into my vehicle and try again

brianfitzgerald, 2021-07-11:
None of the processes are red.

brianfitzgerald, 2021-07-11:
Sorry, I don't think I'll be able to post the dongle ID - once I power it off after a flash it returns to the black screen when I prepare to plug it into my car dongle, unless you have a workaround..

adeebshihadeh, 2021-07-11:
I got your dongle ID, but the logs didn't indicate any problems. I factory reset a device and tried to reproduce but didn't hit any issues. Are you sure there were no errors during flashing?

brianfitzgerald, 2021-07-11:
Here are the logs from my most recent flash.

`                                                   FAILED (remote: 'unknown command')
fastboot: error: Command failed
Sending 'recovery' (14853 KB)                      OKAY [  0.380s]
Writing 'recovery'                                 OKAY [  0.121s]
Finished. Total time: 0.549s
Sending 'boot' (11677 KB)                          OKAY [  0.300s]
Writing 'boot'                                     OKAY [  0.098s]
Finished. Total time: 0.439s
Invalid sparse file format at header magic
Sending sparse 'system' 1/5 (506914 KB)            OKAY [ 12.026s]
Writing 'system'                                   OKAY [  5.644s]
Sending sparse 'system' 2/5 (504707 KB)            OKAY [ 13.051s]
Writing 'system'                                   OKAY [  5.706s]
Sending sparse 'system' 3/5 (514168 KB)            OKAY [ 12.479s]
Writing 'system'                                   OKAY [  5.788s]
Sending sparse 'system' 4/5 (475658 KB)            OKAY [ 11.365s]
Writing 'system'                                   OKAY [  7.244s]
Sending sparse 'system' 5/5 (505884 KB)            OKAY [ 11.792s]
Writing 'system'                                   OKAY [ 45.594s]
Finished. Total time: 134.203s
******** Did you mean to fastboot format this ext4 partition?
Erasing 'userdata'                                 OKAY [  0.066s]
Finished. Total time: 0.087s
mke2fs 1.44.4 (18-Aug-2018)
Creating filesystem with 65536 4k blocks and 65536 inodes
Filesystem UUID: 3ca6c80b-58ba-4c44-bcb4-67996df01a23
Superblock backups stored on blocks:
	32768

Allocating group tables: done
Writing inode tables: done
Creating journal (4096 blocks): done
Writing superblocks and filesystem accounting information: done

Sending 'cache' (120 KB)                           OKAY [  0.040s]
Writing 'cache'                                    OKAY [  0.023s]
Finished. Total time: 0.194s
Rebooting                                          OKAY [  0.008s]
Finished. Total time: 0.008s`

adeebshihadeh, 2021-07-12:
Hmm, I have a suspicion it may be an issue with your sensors. We use the accelerometer and gryo to detect taps while the screen is off. Can you SSH in and run `selfdrive/debug/dump.py sensorEvents` once openpilot is installed (need to see sensord in green in that first pane)? Let it run until you see some output for a few seconds then kill it (Ctrl-C) and paste the output here.

rimas8541, 2021-07-12:
Just read thru this thread. Received my brand new C2 on Saturday. Same exact issue. Starts up, gets to the Comma logo startup screen, then goes black with the screen backlight still on. 
this occurs both plugged into the vehicle using the supplied harness and plugged into my Mac. 
I’ve tried powering off and back on and rebooting. 
Figured another example of same issue might be helpful.

adeebshihadeh, 2021-07-12:
@rimas8541 we are unable to reproduce the issue here and haven't seen other reports of this, so it's unlikely to be a software issue. Contact support@comma.ai for a repllacement

closed 2021-07-14 by adeebshihadeh