commaai/openpilot #21572: Comma 2 Says No Panda

JPFoxxx, 2021-07-13:
**Describe the bug**

**How to reproduce or log data**

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

**Additional context**

Also showing no panda Software??

adeebshihadeh, 2021-07-13:
Our issue tracker is not for support. Check out https://discord.comma.ai for help getting started. If you think you've hit a bug, open a new issue and fill out the whole bug report template.

closed 2021-07-13 by adeebshihadeh

JPFoxxx, 2021-07-13:
Thank you I will return the device.