commaai/openpilot #21586: 2019 Honda Fit EX - Extremely Slow Acceleration

Smurf, 2021-07-14:
**Describe the bug**
As of 7/12/2021 I am experiencing OP failing to accelerate in a timely manner. This results in me losing up to 10 mph of speed before OP begins to accelerate.

Engage OP with or without a lead vehicle. OP fails to maintain the speed cruise was engaged at dropping 5+ mph before acceleration begins. This is greatly exacerbated by any in
[image]
cline. See the attached graph from Torqi_boi on Discord showing desired and real acc being 0.5m/ss apart.

**Device/Version information (please complete the following information):**
 - Device: [Comma 2
 - Dongle ID: 52f3e9ae60c0d886
 - Route: 52f3e9ae60c0d886|2021-07-13--18-17-42
 - Timestamps: 
 - Tue, 13 Jul @ 18:18:57 - Car has no lead vehicle and fails to accel until I hit the + button to increase cruise speed
 - Tue, 13 Jul @ 18:21:14 - Car drops 5mph before I start hammering on the + button to increase the cruise speed
 - Tue, 13 Jul @ 18:23:48 - Car is going up hill and instantly drops speed, say f-it and hit the gas and deactivate OP
 - Version: master-ci - 3ad564bfec
 - Car make/model: Honda Fit 2019

**Additional context**

See attached graph.

HaraldSchafer, 2021-07-23:
Closing and linking to generic issue
https://github.com/commaai/openpilot/issues/21697

closed 2021-07-23 by HaraldSchafer

HaraldSchafer, 2021-08-12:
https://github.com/commaai/openpilot/commit/a8f3e2b3c26f2b617a1c9c94250b44adce861d04

Just merged a potential fix into master. This should hopefully help with some Hondas. I was not able to test some, so any feedback is welcome.