commaai/openpilot #21620: After Update to 0.8.3, the car does not recognize the device; Functions as dashcam only.

rx3123, 2021-07-16:
Describe the bug
Openpilot on Comma 2 has been working for the past year, but after the recent Update, the car does not recognize the device. 

How to reproduce or log data
Drove a few miles, the software is openpilot, but C2 seems to function only as a dashcam

Expected behavior
After the update, the device does not recognize the car.

Device/Version information (please complete the following information):

Device: [Comma Two]
Version: [0.8.3],
Car make/model [Lexus RX 350 F Sport]

Dongle ID  3f584d02a9803d68

pd0wm, 2021-07-16:
Can you make sure your logs are uploaded? The most recent route for that dongle id is from March 27.

Do you have your comma power plugged into the OBD port? This is required for fingerprinting.

rx3123, 2021-07-17:
Hi.
I checked and the OBD port connection was loose...the clip connection is
not very tight. I taped the connector on tight with electrical tape and now
it works. Thanks for your help.
Don

closed 2021-07-17 by adeebshihadeh