commaai/openpilot #21637: UI: fix http request caching

adeebshihadeh, 2021-07-17:


adeebshihadeh, 2021-07-19:
@deanlee want to look into this one? would be nice to also write tests

deanlee, 2021-07-19:
ok

adeebshihadeh, 2021-07-20:
this is for things like the API requests for drive stats that we cache to a param

deanlee, 2021-07-20:
I see... but I think it's still better to use QNetworkConfigurationManager to test internet connection in qt/setup. it's more "realtime" and if we clear nam'scache in HttpRequest, the time of clear/re-init may be more than 300ms on EON

deanlee, 2021-07-20:
> this is for things like the API requests for drive stats that we cache to a param

could it be expressed in a more specific way?

adeebshihadeh, 2021-07-20:
We cache the last successful request in RequestRepeater so that things like the drive stats are populated even without internet. https://github.com/commaai/openpilot/blob/master/selfdrive/ui/qt/request_repeater.cc#L15

deanlee, 2021-07-22:
I still don't quite understand how to fix the request caching. can it  be solved if we don't emit receivedResponse from cache when there is no network?

closed 2021-07-23 by adeebshihadeh

adeebshihadeh, 2021-07-23:
Fixed the caching with #21693, but the prime widget doesn't seem to take advantage of the caching properly.

PR #21693 (2021-07-23): Fix RequestRepeater caching
fixes #21637