commaai/openpilot #21685: simulation:OpenPilot and ScenarioRunner for OpenScenario

jbirtel, 2021-07-22:
**Describe the problem**

I am trying to use the ScenarioRunner of Carla to simulate Scenarios in OpenScenario format. Additionally, I use OpenPilot. The problem occurs, when trying to combine ScenarioRunner by Carla and OpenPilot. 
The ego vehicle which is spawned by OpenPilot is not recognized by the ScenarioRunner/Carla. So how is it possible - if it is actually possible - to specify, that the vehicle, that is spawned when starting OpenPilot, triggers the scenario? Right now OpenPilot kind of runs independently from the Scenario and we get so to say two different scenes (LKA and scenario in Carla). That means, we get an extra ego vehicle for the scenario instead of using the one provided/spawned by OpenPilot. I hope you get my point.

**Expected behavior**

I expect that I can combine OpenPilot and Carla+ScenarioRunner to be able to run scenarios with the given vehicle (of OpenPilot).

**Additional context**

    OS: Ubuntu 18 LTS
    CARLA Version: 0.9.11
    Python version: 3.6
    Version 0.9.11
    OpenPilot: 0.8.4

I am happy about anything, that can help to run the scenarios with the OpenPilot vehicle.

adeebshihadeh, 2021-07-22:
I don't think we ever tested the ScenarioRunner. If you get something working, feel free to open a PR. You can also check out the simulation channel at https://discord.comma.ai to see if anyone else has tried this.

closed 2021-07-22 by adeebshihadeh

jbirtel, 2021-07-29:
So I managed to be able to run the example scenarios, that were provided by the scenario_runner.
Therefore I needed to add the line `vehicle_bp.set_attribute('role_name','hero')` before [this line](https://github.com/commaai/openpilot/blob/d7ab611bb0a616374e28eb4f0cdf46bb0172bb4b/tools/sim/bridge.py#L175), such that the name of the vehicle spawned by OpenPilot is recognized by ScenarioRunner. 
Then, the ScenarioRunner needs to be started like `python3 scenario_runner --scenario <scenario_name> --waitForEgo`. After that, when starting your car in the running bridge.py console, the scenario should be triggered.
I have not yet tried own OpenScenario files.
Actually, this does no harm (at least I think so), so this line could be added by default maybe? (I don't get updates of OP, because I now have a good running version, but maybe for others it is a bit annoying to always add that line via vim once having restarted the docker)

During this and my other runs with OpenPilot, I noticed that the steering ratio/torque that can be applied by the car, not only by the LKA but also by hand, is much lower than the possible torque (by Carla). I tried one scenario to follow a vehicle and there was a ~ 90° turn. The leading vehicle easily managed this with around 20 mph and the OP vehicle couldn't take this with 5 mph, even if the curve is recognized by the system. I mean I know that a LKA can't apply arbitrary steering torques, but is this on purpose ? And is there maybe a way to change that?