commaai/openpilot #21697: Poor longitudinal tuning

HaraldSchafer, 2021-07-23:
Some cars (mostly Hondas) have very poor longitudinal tuning. They undershoot desired acceleration from the planner consistently. This causes the car to never accelerate even if the planner tells it to for 10+ seconds.

These cars need to be retuned so this does not happen. Here is the average difference between desired acceleration (from the longitudinalPlan message) and actual acceleration (from the carState message).

`|    | platform(atleast_50_segs_in_last2_days)   |   mean_planned_minus_actual_acc_at_0 |   segs |
|---:|:------------------------------------------|-------------------------------------:|-------:|
| 18 | TOYOTA HIGHLANDER HYBRID 2020             |                         -0.00778193  |     55 |
| 13 | TOYOTA COROLLA HYBRID TSS2 2019           |                         -0.00495117  |    302 |
| 10 | TOYOTA CAMRY 2021                         |                         -0.00277638  |    180 |
| 20 | TOYOTA PRIUS TSS2 2021                    |                         -0.00216974  |    284 |
| 21 | TOYOTA RAV4 2019                          |                         -0.0017907   |    765 |
| 22 | TOYOTA RAV4 HYBRID 2017                   |                         -0.00167831  |    160 |
| 23 | TOYOTA RAV4 HYBRID 2019                   |                         -0.000704918 |    862 |
| 24 | TOYOTA SIENNA 2018                        |                          0.00185491  |     69 |
| 14 | TOYOTA COROLLA TSS2 2019                  |                          0.00283667  |    477 |
| 12 | TOYOTA COROLLA 2017                       |                          0.00460236  |     60 |
| 16 | TOYOTA HIGHLANDER 2020                    |                          0.00577848  |    204 |
| 19 | TOYOTA PRIUS 2017                         |                          0.0115677   |    162 |
|  2 | HONDA CIVIC 2016                          |                          0.11188     |    143 |
|  7 | HONDA RIDGELINE 2017                      |                          0.148708    |    266 |
|  3 | HONDA FIT 2018                            |                          0.555724    |     52 |`

HaraldSchafer, 2021-07-23:
https://github.com/commaai/openpilot/issues/21697 related

che220, 2021-07-23:
So what parameters should we change to get it to zero?

adidasboy517, 2021-07-23:
Any good guides on how to use plotjuggler with the comma2?

HaraldSchafer, 2021-07-23:
@adidasboy517 
https://github.com/commaai/openpilot/tree/master/tools/plotjuggler

Has a README

@che220 
https://github.com/commaai/openpilot/blob/master/selfdrive/controls/lib/longcontrol.py

This is the longitudinal control loop. It has PID values that are specific to each car/brand.
And then there is some car-specific interface code here
https://github.com/commaai/openpilot/blob/master/selfdrive/car/honda/interface.py
and
https://github.com/commaai/openpilot/blob/master/selfdrive/car/honda/carcontroller.py

I'm not super familiar with it myself, it's likely quite a bit more complicated than just tuning some numbers.

sshane, 2021-07-27:
If those Hondas have gas interceptors, then it's probably due to longcontrol outputting an acceleration request, but the pedal taking in a gas percentage, leading to undershoot as 0 m/s/s desired acceleration means that no gas will be applied. I don't think you can tune this out, you need some way to convert acceleration to gas, per car. Or a general conversion function that works well enough for all with the pedal

HaraldSchafer, 2021-07-27:
@sshane If it's just that, shouldn't the PID be able to deal with it?

sshane, 2021-07-27:
Eventually, but there's always going to be some oscillation with acceleration. If you want to maintain your speed openpilot will request 0 acceleration (which is really 0% gas if using a pedal) so it slows down and then integral will build up and overshoot. It may be less responsive to gas on Hondas though so it may never overshoot (I'm used to Toyota).

grekiki, 2021-07-29:
Sounds like you need a gas feedforward term

HaraldSchafer, 2021-08-04:
Related
https://github.com/commaai/openpilot/issues/21803

HaraldSchafer, 2021-08-04:
Basically the main cause of this issue is that honda nidec accel control goes through a strange function, not the long control loop like it does for all cars. I am working on a PR to fix this. But I can only test on our own Civic. So feedback from other cars would be very welcome.

https://github.com/commaai/openpilot/pull/21845/files

If anyone is willing to do some testing that'd be great. @adidasboy517 @che220 @Smurf

KevinYoung2, 2021-08-05:
Hi HaraldSchafer,

I am brand new to github.  I have the issue on my 2016 Civic.  I would be glad to load you change and test it but I don't know how.  If you can point me to where this is explained (load specified SW) I will do that..

grekiki, 2021-08-05:
@KevinYoung2 https://github.com/commaai/openpilot/wiki/Forks

KevinYoung2, 2021-08-06:
I have looked at the link from grekiki above.  I created the following command:
https://smiskol.com/fork/github.com/commaai/openpilot/pull/21845/files
This seems to start the loading process.  I get the 'Installing Openpilot' screen with the comma and revolving circle.  But it never finishes.  I imagine there is something wrong with my command,  Can anyone tell me what the error is?

Thanks.

Na112786, 2021-08-10:
So what until now My problem hasn't been solved?!
#21803.

sshane, 2021-08-10:
@Na112786 it's generally preferred to consolidate identical issues into one. Don't worry, this one hasn't been closed and your issue was linked.

HaraldSchafer, 2021-08-12:
https://github.com/commaai/openpilot/commit/a8f3e2b3c26f2b617a1c9c94250b44adce861d04

This should hopefully help with some Hondas. I was not able to test some, so any feedback is welcome.

Na112786, 2021-08-12:
Feedback,
Nothing change still same problem..
This my Dongle and serial number.. 
[image]

Na112786, 2021-08-12:
[image]

Na112786, 2021-08-12:
[image]

HaraldSchafer, 2021-08-12:
@Na112786 You are on release, which is the stabled released version of openpilot. The changes I made or only on openpilot master. So you would need to install master to test the fix.

https://github.com/commaai/openpilot/tree/master

Na112786, 2021-08-12:
😅 Firstly, am on release 2
Why u make it complicated, do it on release 2.
Secondly, I don't know how to install master?
Don't send link, just do it for me

HaraldSchafer, 2021-08-12:
That's not how the openpilot release process works. We develop on master and push to release when it is ready. 
https://blog.comma.ai/a-2020-theme-externalization/

If you want to learn how this all works and how to switch to master you can do some reading, perhaps check out out discord https://discord.com/invite/avCJxEX

If you feel this is all too complicated I suggest you wait until fixes come to release.

madcapclover, 2021-08-13:
@HaraldSchafer Tried master this afternoon on my drive in to work. As my car (2018 nidec civic) approached the setpoint speed it would start pumping the brakes repeatedly making it unusable, and I had to disengage. 
For example, if I set it to 120 km/hr, it would get up to 119km/hr and then repeatedly pump the brakes making it unusable to drive with. (Wasn't aggressive but enough that you could feel it and enough that it was not "chill")  

Thanks for taking the time to work out these issues

HaraldSchafer, 2021-08-13:
@madcapclover Can you provide a route or your dongleid? I'll investigate

madcapclover, 2021-08-13:
@HaraldSchafer 41f056ee588d15d0 dongleid, thanks!

Smurf, 2021-08-14:
Looks like this got merged into master-ci today. I tested 0.8.8 master CI on a 2019 Honda Fit EX and can confirm that OP stays within 1mph of the set cruise.

However, when there is a lead car it seems highly reactive. 
52f3e9ae60c0d886|2021-08-13--17-10-34
Shows this behavior towards the end of the drive. I am behind a pickup at around 30mph that is changing speed slightly, OP adjusts a bit much but the follow distance is **much improved!**

HaraldSchafer, 2021-08-14:
@madcapclover I see what you mean in the logs. Looking into it.

@Smurf I see what you mean. It looks like the car actually did what the planner wanted. The lead policy just needs to be a little more chill, not sure if I can fix by next release. Please let me know if you have more feedback and if you think it's a regression over current release. Ty!

Smurf, 2021-08-15:
Got to do some more testing at varied speeds today. At speeds of 40mph+ it seems to handle a lead car much better and it allows much more coasting than previous releases. 

While I was able to feel the surging at low speeds while concentrating on the display I had a passenger today and they couldn't notice.

KevinYoung2, 2021-08-15:
I loaded master-ci today and drove for about 20 miles.  2016 Civic.  It performed very well.  Only issue I observed was very active throttle response when following.  This occurred at 75, 65 and 55.  Maintained distance well, but throttle went idle to moderate power pretty often.

HaraldSchafer, 2021-08-17:
I just pushed some changes that should improve the coasting behavior and the pulsing at high speed when following a lead car.

KevinYoung2, 2021-08-17:
I loaded master-ci today (I assume it had HaraldSchafer's above changes in), and drove for about 50 miles. 2016 Civic.  I think the pulsing when following was better than 2 days ago, but still not as good as the system of 2 months ago.  In all other aspects, system performance was very good.

HaraldSchafer, 2021-08-18:
@KevinYoung2 Yesterday's changes from master have not made it to master-ci yet. Master-ci still needs to be compiled. You can try master or wait until master-ci gets the updates today or tomorrow.

KevinYoung2, 2021-08-18:
@HaraldSchafer OK!  I'm pretty much a neophyte here.  Not sure how to go to master.  Is it the same command only designate master instead of  master-ci?

If not I can just wait until it's in master-ci.  Can you tell me how to know if it's included?

HaraldSchafer, 2021-08-18:
@KevinYoung2 if you are ssh'd in you'd just have to checkout master, init and update submodules and then reboot. It'll take some time to compile.

```
git checkout master 
git submodule update --init --recursive
sudo reboot
```

https://github.com/commaai/openpilot/tree/master-ci

You can just see when master-ci was last updated, which is yesterday, before I merged the changes

madcapclover, 2021-08-18:
@HaraldSchafer Loaded master today and drove to work (100kms), all the pulsing is gone and feels smooth, about 3 car lengths between me and the lead car but overall much better than before, thank you!

adeebshihadeh, 2021-09-02:
@HaraldSchafer should we considered this resolved?

tiki-bug, 2021-09-02:
@adeebshihadeh I was waiting to comment on this until my data was available (routes weren't fully uploading, that should be fixed), but with 0.8.8 my longitudinal control is the worst it's ever been, dangerous in some scenarios.  Still seeing the car not make it up to set speed in a variety of terrains, especially hills.  Lead car behavior is where the biggest problem is though.  It's had sluggish response in the last handful of versions, but with 0.8.8 it takes forever to start speeding up after lead car pulls away, and when coming up on a lead car with maybe a 10-15mph lower speed it attempts to drive right into the lead car with little discernible slowing.  It can see the car fine and even indicates that it's getting way too close.  Overall it feels like a control loop with a super slow response, and this causes the follow space to vary wildly unless the lead car holds constant speed for good amount of time.  I'll follow up with a route and dongle.

-Comma2
-Version: 0.8.8 stock, no pedal
-Car: 2020 Honda Odyssey

belm0, 2021-09-02:
> and when coming up on a lead car with maybe a 10-15mph lower speed it attempts to drive right into the lead car with little discernible slowing. It can see the car fine and even indicates that it's getting way too close. Overall it feels like a control loop with a super slow response

This sounds similar to what I experienced on another model (Freed).  But first, note that the regular Odyssey is not using the revised longitudinal control:
```python
OLD_NIDEC_LONG_CONTROL = set([CAR.ODYSSEY, CAR.ACURA_RDX, CAR.CRV, CAR.HRV])
```
@tiki-bug do you have the `ODYSSEY_CHN`?  Please check the `platform` column in your useradmin routes list.

KevinYoung2, 2021-09-02:
Have driven for 100 miles (2016 civic).  It has behaved very well.  No issues noticed.  I did not observe the same behavior as tiki-bug.

tiki-bug, 2021-09-02:
@belm0 That's a good catch, I had assumed that the long fix was applied across the board for hondas and that it was in 0.8.8 since it went to master a couple weeks ago, I guess only the second one is true?  And to answer your other question, no it's the `ODYSSEY = "HONDA ODYSSEY 2018"` not the `ODYSSEY_CHN`

adeebshihadeh, 2021-09-02:
@tiki-bug This PR applies the new control to the Odyssey. It would be great if you could try it and report any feedback. https://github.com/commaai/openpilot/pull/22104/files

HaraldSchafer, 2021-09-02:
@tiki-bug @belm0 

Integrator windup was causing issues with braking. I tried to fix that in master
https://github.com/commaai/openpilot/commit/ebf2a2279d837feb3ce55a7bc44c7f0053e77f27

I also moved the odyssey to new control. If you guys could tell me if that fixed the issues, that'd be great.
It's on master now, will be in master-ci in a couple hours.

belm0, 2021-09-03:
Thank you.  It may be a few days before I can try it on Freed.

Are you considering patching 0.8.8 once you get confirmation on the Odyssey?

tiki-bug, 2021-09-03:
@HaraldSchafer @adeebshihadeh 
Loaded up master-ci for this morning's commute

Car: 2020 Honda Ody
Version: 0.8.9-master-ci

`0a96f86fcfe35964|2021-09-03--07-28-57`
Timestamp: 7:31:15 to 7:31:50
Short drive, but got some good surging data in some light hills around the max speed mark, shows up real well in AEgo

`96f86fcfe35964|2021-09-03--07-42-14`
Timestamps and details:
7:45:18 to 7:47:00 - Jerkiness from brake/accel/brake/etc. around max speed, esp. after transition to lower set speed, w/ and w/o lead car
7:50:50 to 7:51:10 - delayed sharp brake due to cut-in
7:52:00 to 7:52:30 - decent follow, when the lead car below max speed and making small changes in speed
7:57:34 to 7:57:45 - delayed second brake jab as cut-in leaves lane
8:01:01 to 8:01:31 - Surging accel to keep up with lead
8:01:55 to 8:02:15 - delayed sharp brake due to earlier cut-in

Overall I would say the long is significantly jerkier than previous builds but way safer than 0.8.8 release2. It's far more conservative on the follow distance and seems like it can come up to much slower/stopped cars again without driving through them. The jerky/surging behavior is new, and is enough to make the lateral control unstable in new places on slower roads (take pity on our hondas).  Seems like it doesn't want to coast at the set max speed, probably handles this a little better on the highway with the added drag.

HaraldSchafer, 2021-09-03:
@tiki-bug Definitely looks like some things are wrong. Best I can tell, it is not responding to the command that limits acceleration the way other Honda NIDECs seem to. I can help make your car great, but would need you to test some things to understand how the Odyssey is responding to commands, and why it is different. Best would be if you could message me on discord `Torq_boi#2730`, if you're not on discord :
https://discord.com/invite/avCJxEX

HaraldSchafer, 2021-09-06:
Closing this, most Hondas appear to be fixed. Moving Odyssey issue and discussion to here:
https://github.com/commaai/openpilot/issues/22146

closed 2021-09-06 by HaraldSchafer